dhondta/rpl-attacks #1: Fatal error: local() encountered an error (return code 1) while executing 'convert -delay 10 -loop 0

cybercoder, 2016-10-19:
root@instant-contiki:rpl-attacks>> status
┌Status of opened tasks────────┬──────────────────────────────────────────┐
│       Task      │   Status   │                  Result                  │
├─────────────────┼────────────┼──────────────────────────────────────────┤
│ test[make]      │ SUCCESS    │ No result                                │
│ test[run]       │ PENDING    │ Not defined yet                          │
└─────────────────┴────────────┴──────────────────────────────────────────┘
root@instant-contiki:rpl-attacks>> 
Fatal error: local() encountered an error (return code 1) while executing 'convert -delay 10 -loop 0 network*.png wsn-without-malicious.gif'

dhondta, 2016-11-10:
I'm trying to reproduce the use case that caused the error.
It seems that this issue is bound to a Cooja extension.

Did you get the following error while running the command `user@instant-contiki:rpl-attacks>> run [your_experiment]` ?

> All Cooja extensions could not load.
> 
> To manage Cooja extensions:
> Menu->Settings->Cooja extensions

dhondta, 2016-11-12:
This bug was fixed in the last commit (165).

commit fbddefc3 (2016-11-12): Minor improvements (fixed #1)
closed 2016-11-12 by dhondta