dhondta/rpl-attacks #7: Installation error

flsubaei, 2017-02-01:
I'm running it on Ubuntu 16.04 with contiki 3.0 (not a contiki instant) and python 3.5.2

parallels@ubuntu:~/rpl-attacks$ fab test
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/fabric/main.py", line 672, in main
    docstring, callables, default = load_fabfile(fabfile)
  File "/usr/local/lib/python3.5/dist-packages/fabric/main.py", line 173, in load_fabfile
    imported = importer(os.path.splitext(fabfile)[0])
  File "/home/parallels/rpl-attacks/fabfile.py", line 5, in <module>
    from core.commands import get_commands
  File "/home/parallels/rpl-attacks/core/commands.py", line 19, in <module>
    from core.utils.parser import parsing_chain
  File "/home/parallels/rpl-attacks/core/utils/parser.py", line 120
    except AttributeError:
                         ^
TabError: inconsistent use of tabs and spaces in indentation

flsubaei, 2017-02-01:
solved it
the python version was the issue
I think we have to use python 2.7 because fabric does not support 3.5

closed 2017-02-01 by flsubaei

dhondta, 2017-02-02:
From [flsubaei](https://github.com/flsubaei) :

I'm running it on Ubuntu 16.04 with contiki 3.0 (not a contiki instant) and python 3.5.2

~~~~{.console}
parallels@ubuntu:~/rpl-attacks$ fab test
Traceback (most recent call last):
File "/usr/local/lib/python3.5/dist-packages/fabric/main.py", line 672, in main
docstring, callables, default = load_fabfile(fabfile)
File "/usr/local/lib/python3.5/dist-packages/fabric/main.py", line 173, in load_fabfile
imported = importer(os.path.splitext(fabfile)[0])
File "/home/parallels/rpl-attacks/fabfile.py", line 5, in
from core.commands import get_commands
File "/home/parallels/rpl-attacks/core/commands.py", line 19, in
from core.utils.parser import parsing_chain
File "/home/parallels/rpl-attacks/core/utils/parser.py", line 120
except AttributeError:
^
TabError: inconsistent use of tabs and spaces in indentation
~~~~

dhondta, 2017-02-02:
Thank you very much for signalling this bug ; you spotted a junk tabulation bug !
Now fixed.

commit bc2e2413 (2017-02-02): Fixed a junk tabulation remaining (also fixes #7)