dhondta/rpl-attacks #8: running sample-attacks

flsubaei, 2017-02-02:
I have this error when running a simulation.

Can you also please record a video demo of how to run it after the installation

Thank you

----------------
Type help or ? to list commands.
NB: Spaces are separators ; DO NOT use spaces in arguments !

parallels@ubuntu:rpl-attacks>> prepare sample-attacks
2017-02-02 11:51:59 rpla[21323] INFO CREATING NEW EXPERIMENT CAMPAIGN AT '/home/parallels/Experiments/sample-attacks.json'
Traceback (most recent call last):
  File "main.py", line 7, in <module>
    FrameworkConsole(parallel=True).cmdloop()
  File "/home/parallels/rpl-attacks/core/console.py", line 41, in cmdloop
    super(Console, self).cmdloop()
  File "/usr/lib/python2.7/cmd.py", line 142, in cmdloop
    stop = self.onecmd(line)
  File "/usr/lib/python2.7/cmd.py", line 221, in onecmd
    return func(arg)
  File "/home/parallels/rpl-attacks/core/utils/decorators.py", line 150, in wrapper
    kwargs['path']).run(*args, **kwargs)
KeyError: 'path'

Fatal error: local() encountered an error (return code 1) while executing 'python main.py'

Aborting.

dhondta, 2017-02-03:
I figured it out : this issue was introduced by a side-effect while fixing issue #4 .
I will fix it soon.

closed 2017-02-03 by dhondta

dhondta, 2017-02-03:
Now, `prepare` command succeeds.

You should get :

```
user@instant-contiki:rpl-attacks>> prepare sample-attacks
2017-02-03 14:04:44 rpla[31761] INFO CREATING NEW EXPERIMENT CAMPAIGN AT '/home/user/Experiments/sample-attacks.json'
user@instant-contiki:rpl-attacks>> make_all sample-attacks
2017-02-03 14:04:56 rpla[31761] INFO CREATING EXPERIMENT 'test'
user@instant-contiki:rpl-attacks>> status
┌Status of opened tasks────────┬──────────────────────────────────────────┐
│       Task      │   Status   │                  Result                  │
├─────────────────┼────────────┼──────────────────────────────────────────┤
│ test[make]      │ PENDING    │ Not defined yet                          │
└─────────────────┴────────────┴──────────────────────────────────────────┘
user@instant-contiki:rpl-attacks>> 
user@instant-contiki:rpl-attacks>> 
┌Status of opened tasks────────┬──────────────────────────────────────────┐
│       Task      │   Status   │                  Result                  │
├─────────────────┼────────────┼──────────────────────────────────────────┤
│ test[make]      │ SUCCESS    │ No result                                │
└─────────────────┴────────────┴──────────────────────────────────────────┘
user@instant-contiki:rpl-attacks>> 
```

commit 92bacf11 (2017-02-03): Bugfix for 'prepare' command and command termination (fixed #8 and #9)