dhondta/rpl-attacks #11: Create new building blocks

dhondta, 2017-02-06:
Add building blocks for more attacks, i.e. these described in the taxonomy presented in the [Academical Report](https://github.com/dhondta/rpl-attacks/blob/master/doc/report.pdf).

dhondta, 2018-03-03:
**TODO**: Selective Forwarding Attack

faithphd, 2019-06-11:
hi dhondta I want to implement increased-rank attack and dag inconsistency.
my building block is listed below and I want to ask two questions.
1-) is my building block true for dag-inconsistency and increased-rank? (in my experiments, I could not see too much change in power consumption)
if this building blocks false, what do I need to do in order to implement these attacks?

2-)in my simulations, I want to locate malicious mote as a leaf node (as a child of other motes, not as a child of root node), but in all of my simulations malicious mote became a child node of root node and I want to change the location of malicious mote in order to increase the distance between malicious mote and root node. can I do it? if I can, In order to do that which parameter do I need to change? 
I have changed min_distance_between_motes parameter, but I couldn't change the distance between the root and malicious mote and I could locate malicious mote as a leaf mote.

{
  "dag-inconsistency": {
    "RPL_HDR_OPT_DOWN": 1,
    "RPL_HDR_OPT_RANK_ERR": 1
  },
  "'increased-rank": {
    "RPL_CONF_MIN_HOPRANKINC": 512,
    "rpl-private.h": [
      ["#define RPL_MAX_RANKINC             (7 * RPL_MIN_HOPRANKINC)", "#define RPL_MAX_RANKINC              (35 * RPL_MIN_HOPRANKINC)]
    ]
  }
}

dhondta, 2022-03-29:
@faithphd Hi ! I don't if it is still time to answer your questions, but here are some remarks : 

1. No idea if the blocks are correct, but you should check this against [this reference](https://hal.inria.fr/hal-01207859/document), this excellent paper was a foundation of this project. You can check subsection 3.2.2 out for the DAG Inconsistency Attack for the condition to be satisfied. For the Increased Rank Attack, subsection 3.2.1 explains it.
2. In order to change the location, **from within the framework**, use the `cooja` command with your experiment as the argument. This will open Cooja and you will be able to modify the position of your malicious mote. **If you modify positions of other nodes**, it is necessary that you have used the `cooja` command as, while saving your simulation with the new positions and returning to the framework, it **will align both simulations** (with and without the malicious mote).