dhondta/rpl-attacks #20: Help Wanted -

ycwuaa, 2017-02-07:


dhondta, 2017-02-08:
I fixed a bug the `make` command. Normally, a simulation/campaign can be made without error. `run` is not tested yet.

dhondta, 2017-02-09:
Note that I added an ´update´ command to make it easier to either update Contiki-OS as well as RPL Attacks Framework.

dhondta, 2017-02-11:
I tested on **Contiki-30.x-2370** ; the simulations were **successful**.

```
user@instant-contiki:rpl-attacks>> run_all sample-attack.json
2017-02-11 14:55:34 rpla[10135] INFO PROCESSING EXPERIMENT 'test'
user@instant-contiki:rpl-attacks>> status
┌Status of opened tasks────────┬──────────────────────────────────────────┐
│       Task      │   Status   │                  Result                  │
├─────────────────┼────────────┼──────────────────────────────────────────┤
│ test[run]       │ PENDING    │ Not defined yet                          │
└─────────────────┴────────────┴──────────────────────────────────────────┘
user@instant-contiki:rpl-attacks>> 
user@instant-contiki:rpl-attacks>> 
user@instant-contiki:rpl-attacks>> 
┌Status of opened tasks────────┬──────────────────────────────────────────┐
│       Task      │   Status   │                  Result                  │
├─────────────────┼────────────┼──────────────────────────────────────────┤
│ test[run]       │ SUCCESS    │ No result                                │
└─────────────────┴────────────┴──────────────────────────────────────────┘
user@instant-contiki:rpl-attacks>> versions
2017-02-11 14:57:24 rpla[10135] INFO CHECKING VERSIONS
2017-02-11 14:57:24 rpla[10135] WARNING Contiki-OS: 3.x-2370-ge3d5b74
2017-02-11 14:57:24 rpla[10135] WARNING RPL Attacks Framework: 1.1.2-13-gdbe7986
```

dhondta, 2017-02-11:
I tested on Contiki-3.x-3153 ; the simulations **failed**.

```
user@instant-contiki:rpl-attacks>> run_all sample-attacks.json
2017-02-11 14:32:25 rpla[2200] INFO PROCESSING EXPERIMENT 'test'
user@instant-contiki:rpl-attacks>> status
┌Status of opened tasks────────┬────────────────────────────────────────────┐
│       Task      │   Status   │                   Result                   │
├─────────────────┼────────────┼────────────────────────────────────────────┤
│ test[make]      │ SUCCESS    │  No result                                 │
│ test[run]       │ FAIL       │ ValueError: min() arg is an empty sequence │
└─────────────────┴────────────┴────────────────────────────────────────────┘
user@instant-contiki:rpl-attacks>> versions
2017-02-11 14:35:22 rpla[2200] INFO CHECKING VERSIONS
2017-02-11 14:35:22 rpla[2200] WARNING Contiki-OS: 3.x-3153-g263af6f
2017-02-11 14:35:22 rpla[2200] WARNING RPL Attacks Framework: 1.1.2-13-gdbe7986
```
user@instant-contiki:rpl-attacks>> run_all sample-attacks.json
2017-02-11 14:32:25 rpla[2200] INFO PROCESSING EXPERIMENT 'test'
user@instant-contiki:rpl-attacks>> status
┌Status of opened tasks────────┬────────────────────────────────────────────┐
│       Task      │   Status   │                   Result                   │
├─────────────────┼────────────┼────────────────────────────────────────────┤
│ test[make]      │ SUCCESS    │  No result                                 │
│ test[run]       │ FAIL       │ ValueError: min() arg is an empty sequence │
└─────────────────┴────────────┴────────────────────────────────────────────┘
user@instant-contiki:rpl-attacks>> versions
2017-02-11 14:35:22 rpla[2200] INFO CHECKING VERSIONS
2017-02-11 14:35:22 rpla[2200] WARNING Contiki-OS: 3.x-3153-g263af6f
2017-02-11 14:35:22 rpla[2200] WARNING RPL Attacks Framework: 1.1.2-13-gdbe7986
```

Cooja seems to fail because of a plugin, meaning that the simulations crash very early before making the expected results, thus resulting in an error in the parser.

I must still investigate to fix this compatibility issue.

dhondta, 2017-02-14:
**Diagnostic**: Neither Contiki-OS nor RPL Attacks Framework is the problem ; it seems that Rhino Javascript Engine was replaced in Java 8 and that `importPackage` (present in simulations' `script.js`) is not supported anymore.

**Status**: Working on a compatibility fix...

dhondta, 2017-02-14:
Cooja error is now handled in `run` command ;

```
user@instant-contiki:rpl-attacks>> run_all sample-attacks.json
2017-02-14 12:55:37 rpla[3028] INFO PROCESSING EXPERIMENT 'test'
user@instant-contiki:rpl-attacks>> status
┌Status of opened tasks────────┬──────────────────────────────────────────┐
│       Task      │   Status   │                  Result                  │
├─────────────────┼────────────┼──────────────────────────────────────────┤
│ test[run]       │ PENDING    │ Not defined yet                          │
└─────────────────┴────────────┴──────────────────────────────────────────┘
user@instant-contiki:rpl-attacks>> 2017-02-14 12:55:51 rpla[3036] ERROR Cooja error:
FATAL [Thread-1] (LogScriptEngine.java:317) - Script error:
<eval>:2 ReferenceError: "importPackage" is not defined
	at jdk.nashorn.internal.runtime.ECMAErrors.error(ECMAErrors.java:57)
	at jdk.nashorn.internal.runtime.ECMAErrors.referenceError(ECMAErrors.java:319)
	at jdk.nashorn.internal.runtime.ECMAErrors.referenceError(ECMAErrors.java:291)
	at jdk.nashorn.internal.objects.Global.__noSuchProperty__(Global.java:1441)
	at jdk.nashorn.internal.scripts.Script$Recompilation$1$40$\^eval\_.run(<eval>:2)
	at jdk.nashorn.javaadapters.java.lang.Runnable.run(Unknown Source)
	at org.contikios.cooja.plugins.LogScriptEngine$4.run(LogScriptEngine.java:300)
	at java.lang.Thread.run(Thread.java:745)
2017-02-14 12:55:51 rpla[3036] WARNING Cooja failed to execute ; 'run' interrupted (no parsing done)

┌Status of opened tasks────────┬──────────────────────────────────────────┐
│       Task      │   Status   │                  Result                  │
├─────────────────┼────────────┼──────────────────────────────────────────┤
│ test[run]       │ FAIL       │ Exception: Cooja failed to execute       │
└─────────────────┴────────────┴──────────────────────────────────────────┘
user@instant-contiki:rpl-attacks>> 
```

commit 6318cb0b (2017-02-14): Added compatibility with OpenJDK 8 (fixed #20)
closed 2017-02-14 by dhondta