dhondta/rpl-attacks #23: FAIL: test3_update_cooja_user_properties

dmabm, 2017-03-29:
## Hi I received the following error (P1) when I run (fab test) but I was able to load the RPL Attack Framework. There I was trying to execute given sample test suit but I was not able to do that since there was a another problem (P2) as follows

# **P1:**
================================================================
FAIL: test3_update_cooja_user_properties (tests.setup.Test2CoojaSetup)
> Is Cooja's user properties file correctly adapted ?
----------------------------------------------------------------------
Traceback (most recent call last):
  File "tests/setup.py", line 72, in test3_update_cooja_user_properties
    self.assertIn('[APPS_DIR]/visualizer_screenshot', line)
AssertionError: '[APPS_DIR]/visualizer_screenshot' not found in 'SCRIPTRUNNER_LAST_SCRIPTFILE=/home/user/contiki/regression-tests/02-hello-world/hello-world.js\n'
----------------------------------------------------------------------
Ran 11 tests in 19.101s

FAILED (failures=1)

Warning: local() encountered an error (return code 1) while executing 'python -m unittest -v tests'

# 
**P2:**

user@instant-contiki:rpl-attacks>> run_all sample-attacks
2017-03-28 18:39:06 rpla[10382] INFO PROCESSING EXPERIMENT 'test'
user@instant-contiki:rpl-attacks>> 2017-03-28 18:39:27 rpla[10391] ERROR Cooja error:
FATAL [main] (Cooja.java:3696) - Could not load plugin class: VisualizerScreenshot
2017-03-28 18:39:27 rpla[10391] ERROR Cooja error:
FATAL [Thread-1] (LogScriptEngine.java:317) - Script error:
java.lang.reflect.UndeclaredThrowableException

Please help me to figure out these problems, First problem might be the reason for the second problem
Thanks a bunch
Akalanka

dhondta, 2017-04-01:
Hi,

**Reason**: This is surely due to the fact that I removed the setup class method handling the framework setup in the related unit test. I updated the *Installation* procedure in `README.md`.

**Solution**: Please try the following when applying the *Installation* procedure : at step 4, execute `fab setup` then `fab test` (not only `fab test`).

**Please let me know if it solves the problem** (normally, it should).

closed 2017-04-03 by dhondta

dmabm, 2017-04-03:
Hi,

Yes it worked and thank you so much. 

Best Regards
Akalanka

sonxay, 2017-06-20:
Hi...dmabm
My name is Sonxay , I am PhD student in VietNam.
first of all , i would like to say Thanks for your research RPL attacks
after i read your paper..could i have some question to ask you please.

1.  The file serial.log : it likes got the problem.....like cant  active, cant edit, cant run this file 
2. I would like to improve your research : like i want to simulation more like: throughput , end to end delay and Packet delivery Ratio ....How can I do?
Please help me
hope i can get your answer as soon as
thanks
regard
Sonxay