dhondta/rpl-attacks #24: After installing RPL Attack Framwork "collect-view tool" is not there

dmabm, 2017-03-30:
Hi,

After installing RPL Attack Framwork "collect-view tool" is not appear in the Cooja simulator  (See the following images for Before and after installing the Framework available tools in the Cooja simulator )

See before installing at the very bottom in the drop down menu for "collect-view tool" (No Visualizer Screenshot...)

[image]

See after the installing at the very bottom in the drop down menu for "collect-view tool" (But Visualizer Screenshot.. is added to the menu somewhere in the middle)

[image]

Please let me know how to add "collect-view tool" after installing the RPL Attack Framwork

Thank you very much
Akalanka (dmabm)

dhondta, 2017-04-01:
Hi,

**Reason**: RPL Attacks Framework's setup command adapts Cooja's `.cooja.user.properties` in the following way :
- If the file exists, the line starting with `DEFAULT_PROJECTDIRS=` is adapted so that it loads the custom plugin *Visualizer Screenshot* (which is in fact a class extension of the *Visualizer*).
- If the file does not exist, it is created based on the template provided in the RPL Attacks Framework Git project, which only handles the required plugins (and **NOT the *Collect View* plugin**).

**Solution**: Manually configure the plugin through the GUI according to Cooja's documentation of Contiki.

closed 2017-04-03 by dhondta

dmabm, 2017-04-03:
Hi dhondta,

I was able to Manually configured Collect View plugin with the RPL attack framework and now evey thing works fine. Thank you very much for your help.

Actually I may need you help in the future with my research so could you please let me know your e-mail address

Thanks again
Akalanka