dhondta/rpl-attacks #25: How to see the variance of some other parameters such as CPU and Temperature, while attacks are in progress ?

dmabm, 2017-06-02:
I would like to do an experiment to see the Temperature variance  and CUP utilization with and without the attack on a given topology. Could you please let me know how to enable "**Collect View**" tool to see these parameters and do more network analysis with this framework.

OR if there is any other way to do this without using the "**Collect View**" tool please let me know.

Thank you very much!!!

SwitchIsMine, 2017-06-06:
Trying to get the collect view working also. Will let you know if i have any success..

dhondta, 2017-06-06:
Maybe useful reference : [Cooja-Simulator-Manual.pdf](https://www.researchgate.net/profile/Imed_Romdhani/publication/304572240_Cooja_Simulator_Manual/links/5774fa1a08ae1b18a7df9df2/Cooja-Simulator-Manual.pdf)

dhondta, 2017-06-06:
I think this can be handled in the `script.js` of your experiments. You can edit `templates/experiment/scrip.js` and import Collect View and then log information for further analysis.

PS: Sorry, I don't have time enough now to provide you an example (busy with another project). I suggest you Google on keywords like "`cooja collect view javascript examples`" to find your way but I don't think you will find any efficient manual or tutorial to achieve what you want. You would be very kind if you could report on your testing on this matter...

SwitchIsMine, 2017-06-06:
Thats great thanks, I will let you know how I get on!

dmabm, 2017-06-07:
Thank you very much for your information "dhondta" and "SwitchIsMine"
"SwitchIsMine" Please let me update if you were able to do this with some steps.

Thanks again you both...!!!

closed 2017-06-28 by dhondta