dhondta/rpl-attacks #26: Blackhole attack

SwitchIsMine, 2017-06-17:
Hi,

I am currently trying to develop a new building block for the blackhole attack. What packets in the output are considered to be data plane ? I can only seem to see DIO, DAO and DIS messages. Which I thought were all control plane data..

R

SwitchIsMine, 2017-06-19:
Managed to get this working. I stopped the malicious node from forwarding DAO packets received from its child nodes. I think this is a selective forwarding attack. So now just need to get the node to stop generating its own DAO messages.

dhondta, 2017-06-19:
Please note that, as you can see in [rpl-attacks.json](https://raw.githubusercontent.com/dhondta/rpl-attacks/master/examples/rpl-attacks.json) (which served for the report), the `drop-message` building block was not provided.

In order to completely simulate the black-hole attack, I think you can proceed with either of these two methods :
1. Once the malicious mote has redirected the whole traffic through it, you must just find a way to drop every received packet by modifying the IP-related C files of Contiki. This way, you will act at the data plane without touching the control plane. This solution must be set by tuning your Contiki, especially `core/net/ip*`, as *RPL Attacks* only manages `core/net/rpl`.
2. Otherwise, you can still add a RPL attack building block that just blocks every communication with its parent node, especially the DAO's, like apparently you did.

Note that, in my report, I namely mention the *Greyhole​: by using ​`decreased­-rank` ​ and `​selective­-forwarding`*. You are apparently developing `​selective­-forwarding`. Please share your work so that we can improve *RPL Attacks* through a Pull Request when you will have something fully working, it would be very kind.

SwitchIsMine, 2017-06-20:
Would I be correct in saying that the blackhole attack drops all of the DAO messages whilst the selective forwarding attack drops DAOs which are destined for the root, but still generates its own DAOs for the root. 

I will keep you posted on how the attack is looking, will be using this for my masters project which I have just started. So will be looking to develop blackhole and selective forwarding whilst then creating an application in order to visualize the results which have been obtained through the serial logs. Will be working with rate of DIOs generated, PDR and packet delay.

dhondta, 2017-06-27:
Hi @SwitchIsMine ,

I think you're right but I must confess I don't have time enough now to read relevant documentation again for confirmation.

I think you should consider consulting this reference : [A Taxonomy of Attacks in RPL-based Internet of Things](https://hal.inria.fr/hal-01207859/document). This is a nice paper I used in my academical report and from which I extracted the attacks taxonomy tree and some explanations about the attacks I mentioned and/or implemented within the [building blocks](https://github.com/dhondta/rpl-attacks/blob/master/templates/building-blocks.json).

I would greatly appreciate if you could implement some more building blocks and related attacks for the framework.

I wish you courage for your master thesis. Note that I commiserate with your pain ; I already had the opportunity to complete two master thesis and I'll still have to do one within two years.

Feel free to share your findings and to ask for help or advices.

SwitchIsMine, 2017-06-27:
Hi Dhondta,

That's alright, I realized I wasn't using the echo motes.. so wasnt getting any data packets across the network. Just trying to stop the malicious node from forwarding these on to the root now. Which seems like another impossible feat! I have managed to locate the area in the file in net/ipv6/uip6.c. Sorry to keep pestering you. I must admit, I am not that good with programming. So everything is taking a bit longer than usual. If I want to put this code into the malicious nodes file, then how would i generally go about doing that? 

> 
      UIP_IP_BUF->ttl = UIP_IP_BUF->ttl - 1;
      PRINTF("Forwarding packet to ");
      PRINT6ADDR(&UIP_IP_BUF->destipaddr);
      PRINTF("\n");
      UIP_STAT(++uip_stat.ip.**forwarded**);    //FORWARDED CHANGED TO DROP//
goto send;   // SEND CHANGED TO DROP//
> 

Thanks a lot for the help it is greatly appreciated !

dhondta, 2017-08-18:
Hi @SwitchIsMine ,

Did you manage to solve your problem ?

Here are my remarks/suggestions :

- RPL Attacks Framework is made to act at the RPL level in the network layer (see [report](https://github.com/dhondta/rpl-attacks/raw/master/doc/report.pdf), page 2/14 in the introduction), that is, the control plane of Layer 3 (IP)
- It does not support making changes to the IPv6 implementation
- However, it should be possible to modify sending behavior inside the mote's C code.
- For this purpose, Contiki's "apps" folder should be inspected for examples that could already do something similar or related.

sonxay, 2017-09-25:
Hi....
Could i ask you something

this paper only simulation about hello-flooding attack ?

or all of the type attack in RPL 

thanks

dhondta, 2017-09-26:
Hi,

The academical report addresses, as the theoretical part, a taxonomy of all known attacks on RPL (based on a referenced research paper, considering known attacks at the time of its writing).
As the experimental part, it only addresses attacks that are framed in blue in the taxonomy tree at page 3/14, that is :
- Flooding
- Version Number modification
- Decreased Rank
- Blackhole (without application-level features)

Is it the answer you expected ?

Feel free to ask me other questions if something is unclear...

closed 2018-01-09 by dhondta

wantei, 2020-01-27:
@SwitchIsMine