dhondta/rpl-attacks #27: Simulation Contiki for blackhole

sonxay, 2017-06-19:
hi....
My name is Sonxay , I am PhD student in VietNam.
first of all , i would like to say Thanks for your research RPL attacks
after i read your paper..could i have some question to ask you please.
1. file serial.log : it likes got the problem 
2. I would like to improve your research : like i want to simulation more like: throughput , end to end delay and Packet delivery Ratio ....How can I do?
 Please help me
hope i can get your answer as soon as 
thanks
regard
Sonxay

dhondta, 2017-06-19:
Hey Sonxay, nice to meet you ! I'm glad you appreciate this project ! I'm always very happy to welcome newcomers on this repository.

About `serial.log`, if I correctly understand, you experience some problems with it. Could you post an error trace or something that could indicate an error ?

For your second question, I think this is more or less a duplicate of Issue #25 . I think you should get in contact with [dmabm](https://github.com/dmabm) and [SwitchIsMine](https://github.com/SwitchIsMine) for this matter as, unfortunately, I don't have time to spend on the project at this time. Please read and participate to Issue #25 , this should be an interesting sharing.

dmabm, 2017-06-20:
Hello All,
I am still looking for a solution for this mattes for nearly three weeks. I can simply start collect-view tool with brand new simulation created with cooja simulator directly, but the problem is how to integrate that in to this project. 

The other thing is I was able to load a simulation created using this Framework manually, with Contiki inbuilt cooja simulator. Then I run it manually while opening the collect-view too but that tool did not capture and data form any mote. But it seems serial port is connected to project since some of the serial logs can be observed once collect-view tool is connected.

I am also a graduate student and looking for a experimental test-bed for my research stuff and generate some results to get published. So far the RPL attack framework would be the best environment I fond. (Thanks to dhonta a lot) I am also planning to make some more attacks on this framework and see the behavior of some other parameters.

If I am able to figure it out this issue, surely I will post the solution as soon as possible.

sonxay, 2017-06-20:
Hi...

thanks so much

i will contact them soon 

regard
sonxay
--------------------------------------------
On Tue, 6/20/17, Alex <notifications@github.com> wrote:

 Subject: Re: [dhondta/rpl-attacks] Simulation Contiki for blackhole (#27)
 To: "dhondta/rpl-attacks" <rpl-attacks@noreply.github.com>
 Cc: "sonxay" <xay_22@yahoo.com>, "Author" <author@noreply.github.com>
 Date: Tuesday, June 20, 2017, 3:06 AM
 
 Hey Sonxay, nice to meet you !
 I'm glad you appreciate this project ! I'm always
 very happy to welcome newcomers on this repository.
 About serial.log, if I correctly understand,
 you experience some problems with it. Could you post an
 error trace or something that could indicate an error ?
 For your second question, I think this is more or less a
 duplicate of Issue #25 . I think you should get
 in contact with dmabm and SwitchIsMine for
 this matter as, unfortunately, I don't have time to
 spend on the project at this time. Please read and
 participate to Issue #25 , this should be an
 interesting sharing.
 
 —
 You are
 receiving this because you authored the thread.
 Reply
 to this email directly, view
 it on GitHub, or mute
 the thread.

dhondta, 2017-06-27:
@dmabm : Did you manage to get your things work ? Or do you still have some troubles with the Collect View tool ?

NB: At a first sight, I don't figure out if your troubles are related to the following considerations but keep this in mind ; there was a change in the Javascript engine used by Cooja a few months ago. This means that, while the former version of Cooja used OpenJDK 7 with Rhino JS engine, it now uses OpenJDK 8 with Nashorn, which can cause some errors. For this problem, I added a few lines at the beginning of [script.js](https://github.com/dhondta/rpl-attacks/blob/master/templates/experiment/script.js) which solved the problem while upgrading to OpenJDK 8, but I cannot confirm it is backward-compatible and pointed out that it already caused problems for @sonxay (see Issue #28 ). Maybe this change will still cause other problems...

dmabm, 2017-06-27:
*Hi,*

*I can connect the Collect View tool to project and I can see the serial
logs but tool does not collect the information form sensor network*

*DMABM*

sonxay, 2017-06-28:
Hi...how are you?

can I ask you some question ..?

could you tell me , how do you do the graph like the pics

how many minutes , did you run for the simulation to get the graph like this 

hope i can get your askwer

thanks
regard
Sonxay

[image]

dmabm, 2017-06-28:
Hi Sonxay,

I ran like 2 minus to generate this type of graph. But I did not do it manually and these are automatically drawn or generated by the RPL attack Framework once a particular simulation is executed.  

I have a questions about INT (Interference), what is this INT? and how it is calculated ? and What is this parentage (%) mean ? I need proper interpretations for my questions 

Thank you
DMABM

dhondta, 2017-06-28:
Hi all,

@sonxay : As @dmabm told, the default duration of a simulation is 120 seconds and it automatically draws the graph. It uses a custom *VisualizerScreenshot* plugin in Cooja to get the WSN topology but also draws the DODAG with a graph library and puts the results in the subfolder `results` of your simulation folder. For the power consumption graphs, these are generated using `pyplot` while parsing the collected `powertracker.log`. Note that the duration can be configured through your [`experiment.json`](https://github.com/dhondta/rpl-attacks/blob/master/templates/experiments.json). For more information about configuring this, you can consult [this additional documentation](https://github.com/dhondta/rpl-attacks/blob/master/doc/create-campaigns.md).

@dmabm : For the *INT* metric, I would consider the following :
- Each mote is ON for a reduced period during a defined interval (for low power consumption purpose)
- Each mote has an interference range (configurable through [`experiment.json`](https://github.com/dhondta/rpl-attacks/blob/master/templates/experiments.json)) which influences, during its ON period, the transmission of other motes that are also ON in its interference range
- *INT* thus expresses the contributions of other motes in the range of the considered one
More concretely :
- Looking at [Radio.java](https://github.com/contiki-os/contiki/blob/master/tools/cooja/java/org/contikios/cooja/interfaces/Radio.java), you can see the following code :

```
   /**
   * Returns true if this radio had a connection that was dropped due to interference.
   *
   * @return True if this radio is interfered
   */
public abstract boolean isInterfered();
```

- Looking at [PowerTracker.java](https://github.com/contiki-os/contiki/blob/master/tools/cooja/apps/powertracker/java/PowerTracker.java), you can see the following code :

```
[...]
public class PowerTracker extends VisPlugin {
  [...]
  public static class MoteTracker implements Observer {
    [...]
    public MoteTracker(Mote mote) {
      [...]
      } else if (radio.isInterfered()){
        lastRadioState = RadioState.INTERFERED;
      [...]
    }
    [...]
    public void update() {
      [...]
      /* Radio tx/rx */
      [...]
      } else if (lastRadioState == RadioState.INTERFERED) {
        accumulateRadioIntefered(now - lastUpdateTime);
      }

      /* Await next radio event */
      [...]
      } else if (radio.isInterfered()) {
        lastRadioState = RadioState.INTERFERED;
      [...]
      radioWasOn = radio.isRadioOn();
      lastUpdateTime = now;
    }
    [...]
  }
  [...]
}
```

This shows the control flow for managing the interference state of a mote.

As a consequence, applied to the example of the flooding attack :
- In the situation *without the malicious mote*, everything works fine, that is, with a reduced interference on each mote.
- In the situation *with the malicious mote*, some motes are forced to more often remain ON and are thus more impacted by the interferences of their neighbours (e.g. the malicious mote 11 is impacted by motes 3, 7, 10 and has thus a non-negligible *INT* metric).

dmabm, 2017-06-28:
Hi dhondta,

Thank you very much for your detail explanation. I requested those information since I am tiring to do this analysis in different way to interpret the same result in different way. I am going to count CPU, RX, TX separately and draw graphs manually using EXCEL. To do that I plan to use simulation generated data with different networks. That would be my 1st experiment and later I would like to do more similar experiments with some more different parameters instead of "POWER", in that case I may want to connect collect view tool with this Framework.

Thanks again
DMABM

dhondta, 2017-08-06:
Please refer to #31 .

closed 2017-08-06 by dhondta

dharejo, 2018-02-19:
hi all,
i am huma i am new user of cooja simulator em doing my thesis... dhondta abd dmabm can you both plz help me?
I  want to know how plot the graph of delay, Packet Rx and Tx and energy consumtion?

dharejo, 2018-02-19:
My last question is ... how to run two different motes on different protocols separately? kindly share the link or give me the detailed answer.

dhondta, 2018-02-22:
Hi,

The JavaScript
<https://github.com/dhondta/rpl-attacks/blob/master/templates/experiment/script.js>
associated with a simulation file
<https://github.com/dhondta/rpl-attacks/blob/master/templates/experiment/simulation.csc>
allows to define which metrics are to be recorded. This can be tuned to
your needs for collecting Packet Rx, Tx and energy consumption. Note that
this is already handled when simulating an attack with the framework, which
will produce in your experiment's folder the desired results and charts for
use in a report. The framework also generates a PDF from a Markdow-based
report template
<https://github.com/dhondta/rpl-attacks/blob/master/templates/experiment/report.md>
for which you only have to fill in the remark boxes for commenting the
observations of the results.

Does it help you ? Or would you like more information ?

Greetz,

Alex

dhondta, 2018-02-23:
Hi,

For your question about the protocols, the framework does not allow yet to
handle multiple malicious motes (this is foreseen in an open issue
<https://github.com/dhondta/rpl-attacks/issues/14>) and it could therefore
allow to handle different protocols on different malicious motes.
If your question is about all the different motes (regardless of their
roles), different protocols are not mixed together.
However, I don't really get the sense of your question ; what would be your
use case with different protocols ?

Regards,

Alex

2018-02-23 0:56 GMT+01:00 Alexandre D'Hondt <alexandre.dhondt@gmail.com>:

> Hi,
>
> The JavaScript
> associated with a simulation file
> allows to define which metrics are to be recorded. This can be tuned to
> your needs for collecting Packet Rx, Tx and energy consumption. Note that
> this is already handled when simulating an attack with the framework, which
> will produce in your experiment's folder the desired results and charts for
> use in a report. The framework also generates a PDF from a Markdow-based
> report template
> for which you only have to fill in the remark boxes for commenting the
> observations of the results.
>
> Does it help you ? Or would you like more information ?
>
> Greetz,
>
> Alex
>
>
>> My last question is ... how to run two different motes on different
>> protocols separately? kindly share the link or give me the detailed answer.
>>
>>
>
>
>
> A. D'HONDT, Ir²

dharejo, 2018-02-25:
Thank you so much for your quick reply and help.
i will try it and will let you know if it does not work.
and my second question is i want to plot the graph of packet Rx an Tx,
energy consumption and delay for any three motes (  like cooja mote,
wismote, Z1 mote and micazmote).

-- 

*Ms. Huma Dharejo*
Lecturer
Department of Computing and Technology,
Faculty of Engineering, Science and Technology

*Indus University*
ST-2D, Block 17, Gulshan-e-Iqbal. | Karachi, Pakistan
(92-21) 34801430 + Ext 153
huma@indus.edu.pk
 Follow us: facebook.com/IndusUniversityPK

dharejo, 2018-02-25:
i have performed simulation on sky mote.and i have plot different graphs.
i took five sink nodes
(/home/user/contiki-2.7/examples/ipv6/rpl-collect/udp-sink.c)and 5 sender
nodes  (/home/user/contiki-2.7/examples/ipv6/rpl-collect/udp-sender.c).
So my question is i want to perform same simulation for Z1 mote, ESB Mote,
Micaz Mote, wismote guide me how to do these simulation and find the graphs.

-- 

*Ms. Huma Dharejo*
Lecturer
Department of Computing and Technology,
Faculty of Engineering, Science and Technology

*Indus University*
ST-2D, Block 17, Gulshan-e-Iqbal. | Karachi, Pakistan
(92-21) 34801430 + Ext 153
huma@indus.edu.pk
 Follow us: facebook.com/IndusUniversityPK

abhiverma866, 2019-02-02:
> […](#)
> -- *Ms. Huma Dharejo* Lecturer Department of Computing and Technology, Faculty of Engineering, Science and Technology *Indus University* ST-2D, Block 17, Gulshan-e-Iqbal. | Karachi, Pakistan (92-21) 34801430 + Ext 153 huma@indus.edu.pk Follow us: facebook.com/IndusUniversityPK

simpy set TARGET=z1 for using Z1 mote binary.

abhiverma866, 2019-02-02:
simpy set TARGET=z1 for using Z1 mote binary.

IlhammB, 2024-04-07:
> Hello All, I am still looking for a solution for this mattes for nearly three weeks. I can simply start collect-view tool with brand new simulation created with cooja simulator directly, but the [...]

> Hi Sonxay,
>
> I ran like 2 minus to generate this type of graph. But I did not do it manually and these are automatically drawn or generated by the RPL attack Framework once a particular simulation is [...]

IlhammB, 2024-04-07:
Hello everyone, 
Is there anyone who can help me implement the black hole attack in RPL COOJA by providing resources where I can find the code or things that can help me implement this attack? 
Thank you very much.

researchrinki25, 2024-06-29:
[image]

researchrinki25, 2024-06-29:
hii sir  I have this problem when i run cooja 3.0 by using vmware 17 on windows 10 platform. A few days before i make a simple udp client-server communication on that  when i use collect view to take some mote information but didn't get any  information like didn't show the all node in collect view, diidn't get any power usage of mote ,didn't get any node information and many more please guide me what i do to make a simulation on contiki-3.0 and how to get rid of this problem shown in image

abhiverma866, 2024-07-01:
> Hello everyone, Is there anyone who can help me implement the black hole attack in RPL COOJA by providing resources where I can find the code or things that can help me implement this attack? Thank you very much.

Use these files to implement blackhole attack [Contiki 3.0]
[Blackhole.zip](https://github.com/user-attachments/files/16047368/Blackhole.zip)

abhiverma866, 2024-07-01:
> hii sir I have this problem when i run cooja 3.0 by using vmware 17 on windows 10 platform. A few days before i make a simple udp client-server communication on that when i use collect view to take [...]

Probably you need to recompile Cooja from in case of such errors. I recommend switching to Contiki-NG for better support and features.

far-voyager21, 2025-02-16:
Hi, I am an M.Tech student working on my dissertation, and I am currently simulating RPL attacks (Blackhole, MITM, and DDoS) using the Cooja simulator in Contiki. I am facing difficulties in implementing these attacks, particularly in modifying the RPL protocol and writing the necessary C code files.

I would greatly appreciate any guidance, sample code, or references that could help in successfully simulating these attacks. If there are any existing implementations or related discussions, please let me know. I came across a github repo dhondta but I want the code files of these attacks so that later on I can make some changes according to my project.

Looking forward to your support. Thanks in advance!