dhondta/rpl-attacks #28: problem black hole

sonxay, 2017-06-20:
hi.....I am Sonxay

this is my problem ....it is not have series.log in the example

Did you get the image

[image]

SwitchIsMine, 2017-06-20:
Hi Sonxay,

I had this problem as well. You just need to close the script error dialog box and run the simulation again. Once you do this, the simulation will run as it is supposed to!

R

sonxay, 2017-06-20:
Hi...Sorry
could you help me.

I can not active this file cause I wanna to improve your file to do more simulation like : Throughput , end to end delay ....if i can not active this file , it is mean i can not fix anything 

so could you help me pl

thanks
Sonya

[image]

dhondta, 2017-06-20:
Please try the following :
1. As a first try, just remove the aforementioned lines from the Javascript (that is, lines 1 to 4) and try restarting the simulation.
2. If it does not work, please `apt-get update` and `upgrade` your OS, then in *RPL-Attacks*, launch the `update` command.

sonxay, 2017-06-22:
Hi...i just wanna say ..Thank you so much for your helping 

I wish you to be happy and successful for your life 

take care 

sonxay

dhondta, 2017-06-24:
You're welcome, sonxay, that's a pleasure.
I wish you the same !

closed 2017-06-24 by dhondta

jemmyjell55, 2022-03-28:
@sonxay could u help me in to understand the script