dhondta/rpl-attacks #29: Fatal error: local() encountered an error (return code 1) while executing 'md5sum requirements.txt'

dmabm, 2017-06-21:
Hello All,

I encountered an error as follows when I run "fab test", when I am trying to reinstall the framework with python-2 in a different virtual machine. "fab setup" worked fine. I did not get this problem last time when I was installing the framework with ppython-2 and at that installation worked so fine. (I updated the O/S using "update" comment but still did not work)

Please let me know how to solve this problem. (Screen shot of the problem is attached). 

[image]

dhondta, 2017-06-24:
The problem should be solved. Please ´git pull` and tell me if it works.

dmabm, 2017-06-26:
Hi dhondta,

Thanks for the help and not that problem is solved but after fixing that error, I got following problem when I try to run "fab  test". Please have a look at it.
[image]

Thank you

dhondta, 2017-06-27:
Sorry, I made a stupid mistake (`os.path.exists` instead of `exists` while I use `from os.path import `exists` in the imports)...
This should be solved for now. Could you test and tell me, please ?

dmabm, 2017-06-27:
Hi,

You are fine dhondta & I just tested, but now I am getting the previous error with "fab test" as follows 
**"Fatal error: local() encountered an error (return code 1) while executing 'md5sum requirements.txt'"**
The same error as in previous very 1st post under this issue (Have a look 1st screenshot).
Could you please have look at it again please.
thank you

dmabm, 2017-06-29:
Hi dhondta,
Were you get a chance to have a look at this problem.

Thank you.

dhondta, 2017-07-15:
Hi dmabm,
I was gone for a few days. I'm now working on fixing this problem...

dmabm, 2017-07-19:
Hi dhondta,

No problem, take your time. Once you are done with this fix please update me. These days I am also kind of busy with some other stuff, but I really want to continue my research with this environment. 

I may need your help with the integration part "Collect view tool" with this framework, since I spend lot of time on it but no more progress 

Thanks
Akalanka

dmabm, 2017-08-02:
Hi dhondta,

Just a kind reminder, did you get a chance to have a look at this problem ?

Thanks
Akalanka

dhondta, 2017-08-04:
Hi @dmabm ,

I tested on my side and it seems that the problem should now be fixed.
Please try and let me know.

Greetz.

jomicobasi, 2017-10-05:
Hi, dhondta!
I still have the same error as dmabm:

```
2017-10-05 14:31:57 rpla[2161] INFO TESTING THE FRAMEWORK
[localhost] local: python -m unittest -v tests

Fatal error: local() encountered an error (return code 1) while executing 'md5sum requirements.txt'

================================ Standard error ================================

md5sum: requirements.txt: No such file or directory

================================================================================

Aborting.

Warning: local() encountered an error (return code 1) while executing 'python -m unittest -v tests'

Done.
```
Thanks!

dhondta, 2017-10-06:
This should now be fixed. However, I'm currently unable to test and confirm it.
Could you retry and let me know if the problem is solved, please ?

jomicobasi, 2017-10-06:
@dhondta  thank you for your quick response!
However, another error has arisen :( 

```
2017-10-06 10:52:57 rpla[28032] INFO TESTING THE FRAMEWORK
[localhost] local: python -m unittest -v tests
2017-10-06 10:52:59 rpla[28056] WARNING You are about to loose any custom change made to Contiki-OS ;
2017-10-06 10:53:03 rpla[28056] WARNING Restarting the framework...
python: can't open file 'python -m unittest': [Errno 2] No such file or directory

Warning: local() encountered an error (return code 2) while executing 'python -m unittest -v tests'

Done.
```
Many thanks!

dhondta, 2017-10-07:
Ok @jomicobasi , I think I got it ; the problem should now (really) be fixed and, this time, I tested myself.
I apologize for the inconvenience and thank you for your inputs.

Abhay2doit, 2018-06-15:
HI,
I am encountering the same issue, could you please explain what you did to fix the issue?

dhondta, 2018-06-15:
Hi @Abhay2doit ,

**Problem** (just before [*Third fix to issue #29*](https://github.com/dhondta/rpl-attacks/commit/7000a23a47f75d97cac3a0e332c8d436f13ee9f5) : The error occurs according to the context used ; the `update` command tries to **restart the framework** at the end of its execution.

- While **using** the framework, that is, inside the CLI, the `update` command will then normally restart it.
- While **testing** the framework, the `update` command is called, executing the `restart` function (see `core/common/helpers.py`) with "`python -m unittest`".

**Solution** (with [*Third fix to issue #29*](https://github.com/dhondta/rpl-attacks/commit/7000a23a47f75d97cac3a0e332c8d436f13ee9f5) : I added an `if` to manage the context and avoid using the `restart` function while testing.

**Caveat** : I must argue now that this is a (very) poor solution, just avoiding the problem but definitely not solving it. Indeed, another problem arises ; my bad, I mistakenly reviewed the error in @jomicobasi 's traceback and did not notice that the command issued by the `restart` function seems to provide a call "`python 'python -m unittest'`" (just try it in a terminal and you will get the same error message), hence the error telling that "`python: can't open file 'python -m unittest'`".

I'll publish a fix soon.

Thank you so much for reporting this bug.

dhondta, 2018-06-15:
This fix should have fixed the bug.
Please let me know if you still encounter the error, please.

closed 2018-07-12 by dhondta

oots-1510, 2020-10-13:
@dhondta `Fatal error: local() encountered an error (return code 1) while executing 'pip-sync requirements-dev.txt'` encountered this error while running `./init-workstation.sh ` in local-dev