dhondta/rpl-attacks #30: Why PowerTracker only shows output of Radio ON (%), Radio TX (%), RAadio RX (%) ?

dmabm, 2017-06-28:
Hi All,

I loaded the "test simulation" in the RPL Attack Framework manually by opening cooja using "ant run" and I have few question as follows.

Q1. Why "PowerTracker" does not show "Radio INT" valus (%) ? But after executing same simulation with RPL Attack Framework, under the data folder in the "powertracker.csv" file I can see all the values of 4 parameters (Radio ON (%), Radio TX (%), RAadio RX (%), Radio INT (%).

Q2. What is actual meaning of INT (Interference) and how does it calculated ?

Q3. All 4 parameter values are generated as a presentation (%). what does it mean ? and How does is calculated ?

Q4. How to calculate the CPU time with these parameters?

Q5. That is the different between cooja "Power Tracer" and "Power Tracker" ? Since in build "Power Tracer" attached to the "Collect View" tool give me 4 different parameters such as "CPU, LPM, RX, TX" as output which I can understand very well.  

Please give me some formulas or equations to calculate these parameter. So far I could not find good documentation to solve these problems 

Thank you very much.

dhondta, 2017-06-28:
These questions are partially answered in #27.

dhondta, 2017-06-28:
*Please give me some formulas or equations to calculate these parameter. So far I could not find good documentation to solve these problems.*

I cannot answer it like this. Moreover, answering it would be doing the job in your place as, if I understand correctly, this is part of the implementation of your research, which is not a part of what is implemented yet in the framework. Your questions should be asked on a community forum of the Contiki project.

**Hint**: In order to solve your problem, you should have a close look at the [source code of Contiki and Cooja](https://github.com/contiki-os/contiki). By analyzing the work flow, you will be able to understand which variables you have access to in order to derive metrics.

dmabm, 2017-06-28:
Thank you very much for your valuable information...!!!

closed 2017-07-15 by dhondta