dhondta/rpl-attacks #31: Integration of Collect View tool

dhondta, 2017-08-06:
**Reference**: #27

@dmabm wrote:

*I am going to count CPU, RX, TX separately and draw graphs manually using EXCEL. To do that I plan to use simulation generated data with different networks. That would be my 1st experiment and later I would like to do more similar experiments with some more different parameters instead of "POWER", in that case I may want to connect collect view tool with this Framework.*

musaosman78, 2020-01-04:
dhondta how to write script to collect the node info from collect-view tool in .pcap file

deGhost, 2020-07-02:
Is there a way to use collect-view tool? (I need to access Rtmetric, ETX, Churn)

dhondta, 2022-03-28:
@musaosman78 If you need to collect information during a simulation, you need to edit its `script.js` using the relevant API for the tools you use (i.e. CollectView). Difficult to tell you more as I did not use it yet.