dhondta/rpl-attacks #32: Python problem run_all: TypeError: 'generator' object has no attribute '__getitem__'

renzoe, 2018-01-09:
Hi! First of all thank you for this code.
Very flexible framework.
However, I cannot make it work, I had some issues with pyton dependencies, after solving it, I can make all work , untill I reach 
`run_all sample-attacks`
If I do status I get:
```

┌Status of opened tasks─┬────────────┬──────────────────────────────────────────────────────────────┐
│          Task         │   Status   │                            Result                            │
├───────────────────────┼────────────┼──────────────────────────────────────────────────────────────┤
│ blackhole-attack[run] │ FAIL       │ TypeError: 'generator' object has no attribute '__getitem__' │
│    test[run]          │ FAIL       │ TypeError: 'generator' object has no attribute '__getitem__' │
└───────────────────────┴────────────┴──────────────────────────────────────────────────────────────┘

```

If I run with fab:

```
fab run_all:sample-attacks
2018-01-09 10:00:51 rpla[2553] INFO PROCESSING EXPERIMENT 'test'
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/fabric/main.py", line 762, in main
    *args, **kwargs
  File "/usr/local/lib/python3.6/dist-packages/fabric/tasks.py", line 425, in execute
    results['<local-only>'] = task.run(*args, **new_kwargs)
  File "/usr/local/lib/python3.6/dist-packages/fabric/tasks.py", line 172, in run
    return self.wrapped(*args, **kwargs)
  File "/home/renzo/trabajo/demo-fic/rpl-attacks/core/utils/decorators.py", line 173, in wrapper
    f(*args, **kwargs)
  File "/home/renzo/trabajo/demo-fic/rpl-attacks/core/commands.py", line 558, in run_all
    run(name) if console is None else console.do_run(name)
  File "/home/renzo/trabajo/demo-fic/rpl-attacks/core/utils/decorators.py", line 173, in wrapper
    f(*args, **kwargs)
  File "/home/renzo/trabajo/demo-fic/rpl-attacks/core/commands.py", line 413, in __run
    parsing_chain(sim_path)
  File "/home/renzo/trabajo/demo-fic/rpl-attacks/core/utils/parser.py", line 21, in parsing_chain
    convert_pcap_to_csv(path)
  File "/home/renzo/trabajo/demo-fic/rpl-attacks/core/utils/parser.py", line 55, in convert_pcap_to_csv
    f.write(out)
TypeError: a bytes-like object is required, not 'str'
```

I assume is also w problem with python. I have both pythons. I will keep checking but I will appreciate if you already run into this problem.

On instant contiki 3.0 I also runt with the same problem.

dhondta, 2018-01-09:
@renzoe 

Thank you for this bug report.

**Problem**: String object type in `core/utils/parser.py` at line 53.

**Trigger**: You don't have Tshark install, therefore throwing `OSError` exception of line 52, then setting `out` to a normal string. As line 54 states a `wb` in `open`, using Python 3 throws the exception you received.

renzoe, 2018-01-09:
Hi @dhondta ! Thank you, indeed! Now I installed Tshark by hand (I have runt the -requirelents.txt, apparently is not there?),
Now I got the same mistake as with python3, I commented

line 171 of `core/utils/parser.py` :+1: 

`   """pyplot.legend((p[0] for p in plots), (i.upper() for i in items))     """`

and now It works :) , apparently there is a problem with the iteration:

```
 File "/home/renzo/trabajo/demo-fic/rpl-attacks/core/utils/parser.py", line 171, in draw_power_barchart
    pyplot.legend((p[0] for p in plots), (i.upper() for i in items))
  File "/home/renzo/.local/lib/python3.6/site-packages/matplotlib/pyplot.py", line 3744, in legend
    ret = gca().legend(*args, **kwargs)
  File "/home/renzo/.local/lib/python3.6/site-packages/matplotlib/axes/_axes.py", line 500, in legend
    self.legend_ = mlegend.Legend(self, handles, labels, **kwargs)
  File "/home/renzo/.local/lib/python3.6/site-packages/matplotlib/legend.py", line 583, in __init__
    for label, handle in zip(labels[:], handles[:]):
TypeError: 'generator' object is not subscriptable

```

good enough for me! thanks!

dhondta, 2018-01-09:
@renzoe 

OK, my mistake, I didn't solve the first issue indeed.

Could you please copy-paste the generated crash report ?

A few comments about my design & implementation:
- Unfortunately, due to time shortage, I did not extensively test with Python 3 (and at the time I wrote this framework, this was still 3.5).
- Knowing that problems could happen, especially when working with sub-processes, I added a crash reporting feature (from commit 8236697d96639a06af4c8d2dac11a4638ec537df). This is explained in [Issues Management](https://github.com/dhondta/rpl-attacks#issues-management).

NB: Please note that you could use Vagrant to deploy a VirtualBox instance with the framework installed. This would avoid you all kinds of disagreements like that.

dhondta, 2018-01-09:
@renzoe 

I figured it out ; I used a syntax that resulted in generators instead of tuples and the underlying library (`matplotlib`) causes an error. This is now fixed. Thank you very much for your feedback and I hope you will enjoy playing with my framework !

Note that commenting the line as you did would prevent the power consumption plot from being generated correctly, which would give a weird chart in the report...

closed 2018-01-09 by dhondta

dhondta, 2018-01-09:
@renzoe 

Also, if you enjoy using this framework, it would be very kind to star it. ;-)

And if you want to contribute to new building blocks or features or anything else, do not hesitate...

renzoe, 2018-01-09:
@dhondta  thanks! I am out of my machine at univ, I will send the report tomorrow (but was basically what I copied paste). Thanks again, and next week I willl start using this more, I stared it, and of course If I keep working on this I won t hesitate in contribute to this project!

dhondta, 2018-01-09:
@renzoe You're welcome ! I'm glad to see that this framework can help.

For the report, even if (a priori) I already fixed the problem, it would be kind to show it ; it's an opportunity to see if the crash report feature works as expected and provides sufficient debugging information.

renzoe, 2018-01-10:
Hi, indeed I had plenty of crash reports yesterday, but I erased all but the last two.
I will send one as both point to the same issue:
filename `crash-report-cdec6d12dadad0a8545596e80e44dbdc.txt`

Content:
```

RPL ATTACKS FRAMEWORK - CRASH REPORT
====================================

- For task : test[run]
- Class    : TypeError
- At time  : 2018-01-09 16:44:48

Traceback (most recent call last):
  File "/home/renzo/trabajo/demo-fic/rpl-attacks/core/utils/decorators.py", line 195, in __call__
    return 'SUCCESS', self.f(*args, **kwargs) or 'No result'
  File "/home/renzo/trabajo/demo-fic/rpl-attacks/core/commands.py", line 413, in __run
    parsing_chain(sim_path)
  File "/home/renzo/trabajo/demo-fic/rpl-attacks/core/utils/parser.py", line 24, in parsing_chain
    draw_power_barchart(path)
  File "/home/renzo/trabajo/demo-fic/rpl-attacks/core/utils/parser.py", line 171, in draw_power_barchart
    pyplot.legend((p[0] for p in plots), (i.upper() for i in items))
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/pyplot.py", line 3744, in legend
    ret = gca().legend(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/axes/_axes.py", line 500, in legend
    self.legend_ = mlegend.Legend(self, handles, labels, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/legend.py", line 583, in __init__
    for label, handle in zip(labels[:], handles[:]):
TypeError: 'generator' object has no attribute '__getitem__'
```

dhondta, 2018-01-10:
OK, perfect. Many thanks for the feedback.
Looking at the traceback, I think this should now be fixed. You can use the `update` command in the CLI to get the latest commit.