dhondta/rpl-attacks #33: Version Number Modification Attack Report has wrong DODAG Graph in it?

mueller91, 2018-02-08:
Dear all,
please look at the attached [report.pdf](https://github.com/dhondta/rpl-attacks/files/1707973/report.pdf), which i generated using `fab demo`. To me, it looks like the DODAG graph in the report is the one from the Blackhole Attack? You can see that after the attack, the malicious node is parent of a majority of the surrounding nodes.

dhondta, 2018-02-09:
Hi @mueller91 !

Thank you very much for this feedback. Your interpretation makes sense.
Did you run the simulation multiple times to see what it gives ?

dhondta, 2018-02-10:
Hi @mueller91 

I just started a few simulations and that is kind of weird ; it's just like the *Version Number Modification Attack* had the effect of the *Blackhole Attack*...

This requires further investigation as the ContikiRPL implementation could have been patched and the related attack building blocks could be obsolete.

mueller91, 2018-02-15:
Can you provide me with the version of Contiki that you have been using when it worked? I really find your project useful and would like to help out finding the issue.

dhondta, 2018-02-20:
This is difficult to say ; at the time of the academical project, I did not implement a functionality to check for the versions (of Contiki and the framework itself). So, I did not noted the version when the related attack simulation was run when it gave me the expected result.
Maybe it could be interesting to search for commits in the Contiki repository from June 2016 in order to figure out if significant changes were made to the ContikiRPL implementation. For the time being, unfortunately I do not have time enough to go deeper for checking that.
Please let me know if you find some relevant information about that.

mueller91, 2018-02-20:
I found that building blocks such as `"rpl_recalculate_ranks();", null` were ignored. After fixing this, i reran the experiments, seems to be better now; see the following files: [report_blackhole.pdf](https://github.com/dhondta/rpl-attacks/files/1741097/report.pdf) and [report_version_number.pdf](https://github.com/dhondta/rpl-attacks/files/1741098/report.pdf)

Check out my commit above. It's a quick hack, so feel free to re-implement it yourself. Also, the deletion of the vagrant file was a misclick.

commit 0534bf95 (2018-02-22): Add replacement for 'delete' building blocks. Fixes #33.

Without this, blocks such as '"rpl_recalculate_ranks();", null' are
ignored. This also resolves issue #33, now the graphs are correct in
all the reports.
commit 502b4c6a (2018-02-22): Print replacements from building blocks. Closes #33.
closed 2018-02-21 by mueller91