dhondta/rpl-attacks #34: DIS Protocol in Blackhole Attack PCAP

mueller91, 2018-02-09:
Dear all,

when looking into the output.pcap from the Blackhole attack, i find a lot of packets with the DIS Protocol. Please see the appended image.

DIS is, as far as i can see, the [Distributed Interactive Simulation](https://en.wikipedia.org/wiki/Distributed_Interactive_Simulation), a Protocol used by the Military to perform distributed wargaming.

Can somebody explain what it's used for here? Also, i find it only in PCAP files for the Blackhole attack, never in Hello-Flood or Version-Number attack.

Help is much appreciated.

 [image]

dhondta, 2018-02-10:
Hi @mueller91 ,

What you see in the PCAP is DIS messages, which means DODAG Information Sollicitation. This is a kind of message that sensors send to others to request for a DIO (DAG Information Object) in order to (re)join the network. In this case, it is an indication that the attack works well as these messages are massively sent as the malicious sensor disturbs the WSN.

mueller91, 2018-02-12:
Hi @dhondta,
thank you for taking the time to reply. However, i think that what you describe is not the case.
Consider the following image, which shows a DODAG Information Solicitation (DIS). This is part of the ICMP Message, has Type 155 and Code 0 - as it should.
[image]
Now consider the packets that show up in abundance in the `blackhole attack`, as can be seen in the screenshot in my first post. These packets do not even contain an ICMP message, as shown in the following image.
[image]

The DIS Protocol here is clearly above level 4 of ISO/OSI, since it is above the UDP layer. RPL-DIS, however, is layer 3.

Thus, i don't think what i see here is da DIS from the RPL Protocol. Which makes me wonder, what is it and why is it there? Is it a bug in wireshark (unlikely), or in contiki?

dhondta, 2018-02-20:
My intuition is that this particular case is either a parsing error from Wireshark or simply a kind of ambiguity in this parsing due to similar protocol structures, which could explain the `Unknown PDU` payload we can see in your screenshot.

**Relevant information**:
For the starting point of a proof of this, I would send back to the specifications [RFC](https://www.ietf.org/rfc/rfc768.txt) (DNS) for the DNS datagram and [RFC6550](https://tools.ietf.org/html/rfc6550) (RPL) for the ICMPv6 RPL Control Message (page 30)
 and the DIS message (page 38) structures.

**Observations**:
- In your PCAP, if you take a look at valid ICMPv6 messages, you will see that RPL Control messages are contained in this layer.
- If you compare valid ICMPv6 messages (with DAO and DIO RPL messages) and a DIS one (I mean, a simingly *Distributed Interactive Simulation* protocol message), you will notice that the ICMPv6 layers have significant differences regarding the RPL Options.
- If you take a deeper look at the invalid messages, you will notice that the parsing seems to fail from the `RPLInstanceID`, identifying a `Sender rank` field just after and then parsing a fake UDP layer.
- As it happens, the Blackhole attack relies on the `decreased-rank` building-block which acts on the DAG rank for attracting the traffic to the malicious mote. It the building-block introduces an inconsistency in the RPL Control messages sent by the malicious mote, this could explain why it has an unexpected effect on the final result of the simulation.

**Potential cause**: The Contiki implementation may have changed, therefore causing the related building-block of the attack to be ineffective or faulty, hence causing an inconsistency in the RPL Control messages sent by the malicious mote during the simulation.

**Solution attempt**: I will follow this up with Issue #36 at first glance (in order to search for building-block failures).

mueller91, 2018-02-28:
So, i've had some time to do some investigation. i think there is a bug here different from what you describe above.

**Observation**
- the weird _Distributed Interactive Simulation_ messages appear in both the `malicious` and `without-malicious` pcap files from blackhole (see [without-malicious-pcap.zip](https://github.com/dhondta/rpl-attacks/files/1766200/pcap.zip) )
- they originate not only from the malicious mote, but from all motes
- they are send to more than one mote

If indeed this had to do with the attack, we would expect to see it only in the `malicious` data.

closed 2018-03-21 by dhondta