dhondta/rpl-attacks #35: selective forwording attack

yss8166, 2018-02-19:
hi,

i have sent you any email regarding selective forwarding attack . can you please check it

kurban2015, 2019-08-15:
Dear All,
I am MSc CS student and i need RPL Attacks source code based on Contiki/Cooja simulaor.
I have to sumbit my thesis and i am unable to find source code files of RPL attacks.
Please help me to find out code files. Its an honor for me if anyone please give me source code files.

Regards,
Muhammad Naeem
Email: kurban2015@gmail.com

dhondta, 2019-08-15:
Hi @kurban2015 
I'm not sure to fully understand your request ; I suppose you mean the source code files related to the RPL library of Contiki ?
I suggest you to check this [documentation page](https://rpl-attacks.readthedocs.io/en/latest/building-blocks/) to get started with the JSON configuration of a simulation campaign using the RPL-Attacks framework for tuning ContikiRPL. In particular, the source code files of RPL are located at `[CONTIKI_FOLDER]/core/net/rpl/`.
Does it answer your question ?

kurban2015, 2019-08-15:
Hi Alex,
I just wormhole and grayhole attack source code with. csc extension files
which i can run in contiki/Cooja simulator.
Please send me source code files of RPL attack.

dhondta, 2019-08-15:
@kurban2015 	What do you mean when you say "*source code files*" ?...

kurban2015, 2019-08-15:
I mean Cooja simulator files like with
.csc extension

dhondta, 2019-08-15:
@kurban2015  Every related files is saved in your `~/Experiments` folder under the name of your experiment, including `.csc` and its `.js`, and so forth.

dhondta, 2019-08-16:
@kurban2015 OK, my mistake, I see what you mean. Attacks are implemented with the building blocks and used in the experiments to generate `.csc` files. If you don't find the wormhole and grayhole attacks in the building blocks, then it means that there aren't developed yet.
So, to answer your previous question : I don't have any `.csc` file for these attacks as the related building blocks don't exist yet.

truptipithva, 2021-04-09:
respected sir,
I'm from Gujarat technological university in Ahmedabad Gujarat.
I studied for a master's in cybersecurity last semester.
I want to implement DDOS Attack in the cooja simulator in Contiki OS.
I have no idea about how to write code to implement an attack on a
cooja simulator
and how to find the packet delivery ratio and the threshold for a particular
network. how to find the threshold for a particular network.
If possible then please let me know for further discussion.

Carla-33, 2022-01-28:
Hi All,
II need AODV Attacks code based on scikit-learn simulator using Naive Bayes classifiers. The kinds attack that I have been used are blackhole and selective forwarding. 
Please help me to find out code. Its an honor for me if anyone please give me source code files.

Regards,

truptipithva, 2022-01-29:
Sure .
If I have solution then I will definitely inform you soon.

Carla-33, 2022-01-30:
> Sure . If I have solution then I will definitely inform you soon.
> […](#)

Okay, thank a lot.

far-voyager21, 2025-02-16:
Hi, I’m an M.Tech student working on my dissertation, and I’m currently simulating RPL attacks (Blackhole, MITM, and DDoS) using the Cooja simulator. I’m facing some challenges in implementing these attacks, especially with the C code files for Contiki. If you have any relevant code files, references, or guidance on modifying the existing RPL protocol for these attacks, I would really appreciate your help. Any insights or resources would be a great support for my research.

Thanks in advance!