dhondta/rpl-attacks #40: Installation question

whiteysy, 2018-12-09:
Hi,
   After installing vagrant and virtualbox, which box should i select from the web " http://www.vagrantbox.es/" to add the vagrant box? 
   Look forward to your reply.

whiteysy, 2018-12-10:
OK! Thanks for your reply!

whiteysy, 2018-12-10:
Hi,
I have installed the box successfully and enter the machine by ssh. But I can not find lines staring with "config.proxy" in the Vagrantfile after running the command "vagrant plugin install vagrant-proxyconf". And I want to know how to use the rpl-attacks simulator? What relationship do they have between simulator and the box machine?
Look forward to your reply! 
[image]

holyrocklee, 2018-12-10:
Hello ，dhondta :
        I have some questions about installation. My environment is VMwareworkstation+contiki3.0. Then I want to simulate blackhoel attacks in RPL. How to reailze the model in your repository? According to the README.md, I  tried to install it in my contiki3.0 but failed at last.
      I don't know if my system still does not meet the requirements. It must be noted that I can implement RPL topologies。
      I'm looking forward to your reply!

dhondta, 2018-12-10:
Hi @whiteysy 

The [`Vagrantfile` template](https://github.com/dhondta/rpl-attacks/blob/master/Vagrantfile) mentions the related lines in comment. You can just use and tune these to the proxy of your environment in your `Vagrantfile`.
The RPL Attacks Framework box is an Ubuntu 18.04 Desktop, ready to use. You can just start it and start playing with the framework with the shortcuts defined on the desktop.

I hope this answers your questions.

dhondta, 2018-12-10:
Hi @holyrocklee 

Could you please provide more details about the failure for your installation ? Could you also provide some details about your settings (distribution, Java version, ...) ? Do you use InstantContiki or do you try to install everything manually ? If so, I would advise you to use the Vagrant box, it will be really easier. Please note that, if you're trying to use Java >8, it will fail as the current version of Cooja is only supported by Java 8.

For the blackhole attack, it is already implemented in [`rpl-attacks.json`](https://github.com/dhondta/rpl-attacks/blob/master/examples/rpl-attacks.json), except that the `drop-messages` building-block is not ready yet. You can deploy this case by using the `demo` command (note that it can take a while to make every experiment run, so please be patient).

What do you mean when you tell that you can implement RPL topologies ? Would you mind adding new WSN topologies for complementing the [current ones](https://github.com/dhondta/rpl-attacks/blob/master/core/common/wsngenerator.py) ?

whiteysy, 2018-12-11:
Thanks for your reply and I will try it now.

SUNvega, 2018-12-11:
excuse me，My system version is contiki-3.0. After I opened it, there were only 2GB of memory left. How should I install the 3.66GB virtualbox in the cloud  you provided .Can contiki only use 10GB?

SUNvega, 2018-12-12:
[image]
Excuse me, how can I solve this problem

saideepd, 2018-12-12:
> [image]
> Excuse me, how can I solve this problem

Try these steps for installation: [Vagrant Install & Setup](https://github.com/saideepd/rpl-attacks/blob/master/VAGRANT%20INSTALLATION%20%26%20SETUP%20INSTRUCTIONS.md)

saideepd, 2018-12-12:
> excuse me，My system version is contiki-3.0. After I opened it, there were only 2GB of memory left. How should I install the 3.66GB virtualbox in the cloud you provided .Can contiki only use 10GB?

If possible, expand the Storage space of your Virtual Machine, else try any other Linux Distro.
Also this [Link](https://itsfoss.com/free-up-space-ubuntu-linux/) can be helpful to make some space.