dhondta/rpl-attacks #41: Adding delay in the start of malicious mote.

abhiverma866, 2019-01-30:
I want to add delay in the node startup of malicious mote. My aim is to make a mote send its first DIS after certain period of time. Like in case of hello-flood the DIS_START_DELAY is set to 0. However, when i set delay to 60, it rarely makes any difference. How can i do this. 
[image]
[image]
[image]

dhondta, 2019-12-01:
Hi @abhiverma866 
I apologize for the huge delay.
After a bit of testing, it appears that, whatever I set for the related constant, in practice the delay always seems to be 0.
Namely, I tried a few times to adapt the `malicious-[...].c` files at [`rpl-attacks/templates/experiment/motes`](https://github.com/dhondta/rpl-attacks/tree/master/templates/experiment/motes) to make the malicious mote compile with the option set to multiple test values, but `rpl.log` always shows DIS sent by the malicious mote from the very beginning of the experiment.
I fear this issue comes from Contiki.

abhiverma866, 2019-12-01:
Thank you for your reply. Well the issue was resolved by simply adding a stimer after PROCESS_BEGIN.

closed 2020-05-26 by dhondta