dhondta/rpl-attacks #43: when i try to make a simulation i get cooja error(Cooja error : could not load plugin class: VisualizerScreenshot)

faithphd, 2019-04-02:
[image]

Hi everyone, first of all i would like to say thank you to the owners of this framework.
And i want to ask a question about rpl-attack-framework.

i installed rpl-attack-framework and after that i tried to make a simulation.
i ran prepare sample-attacks command successfuly
i ran make_all sample-attacks command successfuly 
but when i tried to run run_all sample-attacks command i got a cooja error - could not load plugin class: visualizerscreenshot

you can see the all details in attached png file.
if anyone can  help me i will be very happy.
thank you

dhondta, 2019-04-04:
Hi @faithphd ,
Thank you for mentioning this issue.
Are you using the virtual appliance provided with the `Vagrantfile` ?
What is your configuration ?
Please provide the outputs of these commands :
- `uname -a`
- `java --version`
Did you run the `setup` command before trying `run` ?

faithphd, 2019-04-04:
hi dhonta,

I am using the virtual appliance provided with the Vagrantfile.

the commands that I ran in my first experiment :
*prepare sample-attacks
*make_all sample-attacks
*run_all sample attacks

the commands that I ran in my second experiment:
*setup
*prepare sample-attacks
*make_all sample-attacks
*run_all sample attacks

both in my first experiment and the second experiment i got the same errors(which are listed in the attached png file above)

my configuration details are listed below:

user@rpl-attacks-framework:~/rpl-attacks$ java -version
openjdk version "1.8.0_181"
OpenJDK Runtime Environment (build 1.8.0_181-8u181-b13-1ubuntu0.18.04.1-b13)
OpenJDK 64-Bit Server VM (build 25.181-b13, mixed mode)

user@rpl-attacks-framework:~/rpl-attacks$ uname -a
Linux rpl-attacks-framework 4.15.0-38-generic #41-Ubuntu SMP Wed Oct 10 10:59:38 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
user@rpl-attacks-framework:~/rpl-attacks$

dhondta, 2019-04-10:
OK, I think I figured out your problem ;

**Explanation**: You're possibly running the box as it was when I boxed it for Vagrant. It may still contain a bug I fixed later without re-boxing the framework (because this is a time-consuming process).

**Check**: If the command `versions` gives something less than `1.3.0[...]`, you are very likely to have such bug in your appliance.

**Solution**: Execute the command `update` to fix it. Then retry your experiments.

Please tell me if it solved your issue...

faithphd, 2019-04-10:
thank you dhonta,
but when I run update command I get this errors. you can see the errors from screenshot.
[image]

faithphd, 2019-04-10:
hi dhonta , I run update command 
I didnt get any error when i execute update command but it runs only 10 or 15 seconds (what is the average time for update?)
after that the framework restarted

I still get the same errors.(Command `run_all` Cooja error (`could not load plugin class: VisualizerScreenshot`)

dhondta, 2019-04-10:
OK, I will take a look at this deeper.
`VisualizerScreenshot` is my custom plugin for Cooja made to collect screenshots from the builtin plugin `Visualizer`. If something fails with this, it could be because of a regression bug in Java if it was updated/upgraded or because something fails to copy/modify during the execution of the `setup` command inside the framework.
Did you try to update/upgrade Java in your VM ?

dhondta, 2019-04-11:
Hi @faithphd ,

Weird... I just tried and it works fine for me.
My sequence of actions :
- `vagrant up`
- Then, inside the VM : 
  - `prepare sample-attacks`
  - `make_all sample-attacks.json`
  - `run_all sample-attacks.json`

<details>
<summary>`run_all`</summary>
[image]
</details>

<details>
<summary>Results</summary>
[image]
</details>

dhondta, 2019-04-11:
OK, for now, could you try this sequence and post the trace, please ?
- `loglevel debug`
- `update`
- `run_all`

**Note**:
- I didn't see any prompt for sudo in your last screenshot of `setup`, it should not happen with the default VM.
- If we don't manage to get it working, it could be useful either to reinstall the VM with the `Vagrantfile` or, inside your current VM, to remove `rpl-attacks` folder and reclone the Git repo.

closed 2019-08-14 by dhondta