dhondta/rpl-attacks #44: Problem in running Simulation

1amogh3, 2019-04-03:
prepare and make_all commands work properly but when run_all command is run the following error is shown in cooja gui
```
org.contikios.cooja.Cooja$PluginConstructionException: Construction error for tool of class: org.contikios.cooja.plugins.ScriptRunner
	at org.contikios.cooja.Cooja.startPlugin(Cooja.java:1876)
	at org.contikios.cooja.Cooja.tryStartPlugin(Cooja.java:1772)
	at org.contikios.cooja.Cooja.setPluginsConfigXML(Cooja.java:3706)
	at org.contikios.cooja.Cooja.loadSimulationConfig(Cooja.java:3460)
	at org.contikios.cooja.Cooja.loadSimulationConfig(Cooja.java:3380)
	at org.contikios.cooja.Cooja.doLoadConfig(Cooja.java:2408)
	at org.contikios.cooja.Cooja.main(Cooja.java:3243)
Caused by: java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:488)
	at org.contikios.cooja.Cooja.startPlugin(Cooja.java:1856)
	... 6 more
Caused by: java.lang.ClassCastException: java.desktop/javax.swing.text.AbstractDocument$DefaultDocumentEventUndoableWrapper cannot be cast to java.desktop/javax.swing.text.AbstractDocument$DefaultDocumentEvent
	at jsyntaxpane.CompoundUndoMan.undoableEditHappened(CompoundUndoMan.java:63)
	at java.desktop/javax.swing.text.AbstractDocument.fireUndoableEditUpdate(AbstractDocument.java:293)
	at java.desktop/javax.swing.text.AbstractDocument.handleInsertString(AbstractDocument.java:761)
	at java.desktop/javax.swing.text.AbstractDocument.insertString(AbstractDocument.java:716)
	at java.desktop/javax.swing.text.PlainDocument.insertString(PlainDocument.java:131)
	at java.desktop/javax.swing.text.DefaultEditorKit.read(DefaultEditorKit.java:274)
	at java.desktop/javax.swing.JEditorPane.setText(JEditorPane.java:1428)
	at org.contikios.cooja.plugins.ScriptRunner.updateScript(ScriptRunner.java:619)
	at org.contikios.cooja.plugins.ScriptRunner.<init>(ScriptRunner.java:276)
	... 11 more
```

dhondta, 2019-04-04:
Hi @1amogh3 ,
Thank you for mentioning this issue.
Are you using the virtual appliance provided with the `Vagrantfile` ?
What is your configuration ?
Please provide the outputs of these commands :
- `uname -a`
- `java --version`

dhondta, 2019-04-10:
This issue may be similar to #43  ; please ensure you ran `update` before trying your experiments in the framework...

closed 2019-05-02 by dhondta