dhondta/rpl-attacks #45: Error while using 'make_all' command

jalabano, 2019-05-01:
Hi, 
I have a problem while using 'make_all' command. What can be the reason for that error ? I cannot figure it out. Thanks in advance. 

[image]
[image]

dhondta, 2019-05-02:
Hi @jalabano , thank you for reporting this.
Could you provide the crash report that should have been generated in your experiment folder, please ?

jalabano, 2019-05-03:
Actually there is no crash report in the Experiments folder. I think I got an error before creating the crash report.

jalabano, 2019-05-05:
Hi again, 
After reinstalling InstantContiki 3.0 and trying the framework, now I have a different problem. I used "prepare" and "make_all" commands successfully however I got an error after using "run_all" command. After this command, Cooja simulator startup screen showed and when I closed it, I got the following error. Do you have any idea related to this situation? 

[image]
[image]

dhondta, 2019-05-06:
Hi @jalabano 

For the **first problem**, it seems to be **related to Contiki** itself. You may miss some C librairies to compile the motes, therefore causing the command failure error. Also, I figured out that `make_crash_report` is only called for commands decorated with a `CommandMonitor` which is used with multiprocessing (disabled by default at this time). So, this is something I will fix soon.

For the **second problem**, the `run` command starts Cooja to actually run the simulation and the error you got is raised because you closed the simulator before it ends its computation, therefore preventing it to create the log file `powertracker.log` (and surely other ones) in experiment's folder. That's something ugly ; with the `setup` command, I modify the startup settings of Cooja to run it in hidden mode but sadly, it still displays a window (as in your first screenshot). That's something that, as far as I know, I can't fix. So please **do not close Cooja's window** when you see it open as it runs the simulation, producing data and log files.

As you do not seem to use the virtual appliance I provide through the Vagrant Cloud yet, I would suggest you to start using it. It may fix the issue.

closed 2019-08-14 by dhondta