dhondta/rpl-attacks #46: TypeError during script run

abhiverma866, 2019-05-06:
While trying to run the below script i am getting a type error. Possible solutions ??

```
TIMEOUT(300000, log.log("Performance Calculation" + "\n"));

packetsReceived= new Array();
packetsSent = new Array();
timeReceived = new Array();
timeSent = new Array();
count = new Array();
nodeCount = 10;
//data_length = 23;
senderID = 0;
receiverID = 0;
PDR=0;
//e_count = 0;

for(i = 1; i <= nodeCount; i++){
    packetsReceived[i] = 0;
    packetsSent[i] = 0;
    timeSent[i] = 0;
    timeReceived[i] = 0;
}

/*
DATA send to 1 'Hello 1'
Data is Hello 1 from the client
DATA recv 'Hello 1 from the client' from 5
*/

while(1){
    YIELD();
    msgArray = msg.split(' ');
    if(msgArray[1].equals("send")){  
        if(msgArray.length == 6){
            // sent packet
            senderID = parseInt(msgArray[3]);
            packetsSent[senderID]++;
            timeSent[senderID] = time;
        }
    }
    if(msgArray[1].equals("recv")){
        if(msgArray.length == 9){
            receiverID = parseInt(msgArray[8]);
            packetsReceived[receiverID]++;
            timeReceived[receiverID] = time;
            log.log("receiverID " + receiverID + " PacketReceived= " +packetsReceived[receiverID] + "\n");
                if (timeReceived[receiverID] > 0){
                    count[receiverID]++
                }       
            totalReceived = totalSent = 0;
            totalclient=0;
            for(i = 1; i <= nodeCount; i++){
                totalclient++;
                totalReceived += packetsReceived[i];
                totalSent += packetsSent[i];
                log.log("MoteID= " + i + " ReceivedPackets= " + packetsReceived[i] + " SendingPackets= " + packetsSent[i] + "\n");
            }

            log.log(" Generated Packets " + totalSent + "\n");
            log.log(" ReceivedPackets " + totalReceived + "\n");
            PDR=(totalReceived / totalSent) * 100
            log.log("Packet Delivery Ratio" + PDR + "\n");
        }
    }   
}
```

Java stack trace:
<eval>:39 TypeError: Cannot read property "equals" from undefined
	at jdk.nashorn.internal.runtime.ECMAErrors.error(ECMAErrors.java:57)
	at jdk.nashorn.internal.runtime.ECMAErrors.typeError(ECMAErrors.java:213)
	at jdk.nashorn.internal.runtime.ECMAErrors.typeError(ECMAErrors.java:185)
	at jdk.nashorn.internal.runtime.ECMAErrors.typeError(ECMAErrors.java:172)
	at jdk.nashorn.internal.runtime.Undefined.get(Undefined.java:157)
	at jdk.nashorn.internal.scripts.Script$Recompilation$17$40$\^eval\_.run(<eval>:39)
	at jdk.nashorn.javaadapters.java.lang.Runnable.run(Unknown Source)
	at org.contikios.cooja.plugins.LogScriptEngine$4.run(LogScriptEngine.java:300)
	at java.lang.Thread.run(Thread.java:748)

dhondta, 2019-05-06:
Hi @abhiverma866 

I think this could be a compatibility problem due to the replacement of Rhino JS Engine with Nashorn since OpenJDK8. Please try prepending the following code to your script and retry :

```
try { load("nashorn:mozilla_compat.js"); } catch(e) {}
```

abhiverma866, 2019-05-07:
[https://github.com/contiki-ng/contiki-ng/issues/949#issuecomment-489707019](url)

abhiverma866, 2019-05-07:
Reason for that error is that after you execute msgArray = msg.split(' ');, the length of msgArray may be less than 2 (if the string does not contain any spaces). In that case msgArray[1] will be undefined. So i updated my code to handle that special case.

[https://github.com/contiki-ng/contiki-ng/issues/949](url)

closed 2019-05-07 by abhiverma866