dhondta/rpl-attacks #48: Experiment time can not be set  long

chanseng, 2019-09-29:
Dear all,
When I set duration to 1200 sec（or longer)  (The default is 120 sec) in the experiment json file,The experiment will go to half and then no response (usually the without-malicious pcap file is recorded to about 11M).
Has anyone solved this problem? 
Any help is much appreciated.

dhondta, 2019-11-20:
Hi @chanseng 
Thank you for posting this issue.
I never tried with such a long experiment time but I guess it could be something related to the Cooja tool itself. Therefore, I think this question should be asked in the issues of the repository of [Contiki OS](https://github.com/contiki-os/contiki).

chanseng, 2019-11-22:
HI@dhondta
Thank you for replying my question.
I tried to increase the jvm of the cooja,and it worked for 800sec.Maybe 1200sec need more jvm :-).

dhondta, 2019-11-30:
Hi @chanseng 
Thank you very much for this information !

closed 2019-11-30 by dhondta