dhondta/rpl-attacks #53: Error during simulation: TypeError Cannot set property "resetViewport" of null

deGhost, 2020-06-15:
Hello,
When using the commands specified in the presentation under the doc folder:
**make test
cooja test (this already launches cooja simulator)
run test (opens a new cooja window)**
Nothing starts, so I started the simulation manually: simulation->start simulation. After exactly time 00.02.801, the script error message is displayed:
[image]

I really need to use the framework and I've tried many things but it's not working. Also, I thought I should add this here when I use the command" "demo" the console is stuck at PROCESSING EXPERIMENT of blackhole-attack
[image]

My SYSTEM specifications I'm using: 
Contiki 3.0 with ubuntu 16.04
Python 3.5
Openjdk 8
Hope to get some help as soon as possible, thank you!

dhondta, 2020-06-17:
Hi @deGhost 

Thanks for reporting.

For the demo, please beware that it may take a while ; simulations are by default set to a duration of 300s, approximately speed up to ~120s. Did you wait long enough for the run to complete ?

Also, I provide a Vagrant box (sorry, it's huge, with an Ubuntu Desktop of ~4GB) but it is far more convenient for avoiding such errors. This way, you work with the exact same version as I do. When I see the error above, '`Cannot set property "resetViewport" of null`', this may come from various issues, namely if you use a different version of Cooja. This may mean you have a different version of Contiki. I suggest you to `vagrant up` to get the appliance with the tested project on it.

Please let me know if it solves your issue.

deGhost, 2020-06-18:
@dhondta Thanks for taking the time to reply.
I'm using contiki on a virtual machine, I followed the installation setup to use vagrant and thus use the RPL-attacks framework on a nested virtual machine, I've tried all the solutions proposed on stackoverflow which are related to Vagrant but It did not work, so I ended up installing all the system requirements, the packages for python and run the framework using python3.5 main.py to run it. 

I did wait for more than 30 minutes on many occasions to allow it to finish the demo but nothing happens.

Because I'm more interested in the pcap files of each attack, I am now using the simulation file for each attack " with malicious nodes " to generate the pcap files and when the script error occurs I use "step" and start the simulation.

If I had more time I would dig into the cause of this error sorry!

closed 2020-08-05 by dhondta