dhondta/rpl-attacks #54: problem with fab test

yakoubi5103, 2020-08-10:
when i execute the command fab test i have this problem:
Fatal error: local() encountered an error (return code 1) while executing 'git checkout master'

================================================ Standard error ================================================

error: Your local changes to the following files would be overwritten by checkout:
	core/net/ip/tcpip.c
	core/net/ipv6/uip-ds6-route.c
	core/net/rpl/rpl-conf.h
	core/net/rpl/rpl-dag.c
	core/net/rpl/rpl-icmp6.c
	core/net/rpl/rpl-timers.c
	core/net/rpl/rpl.c
	examples/ipv6/rpl-udp/Makefile
	examples/ipv6/rpl-udp/udp-client.c
Please, commit your changes or stash them before you can switch branches.
Aborting

================================================================================================================

Aborting.

yakoubi5103, 2020-08-10:
there is some one who can help me

dhondta, 2020-08-11:
Hi @yakoubi5103 !

Thank you for mentioning this.
It occurs when the [setup test](https://github.com/dhondta/rpl-attacks/blob/master/tests/setup.py) is started. The [`update` command](https://github.com/dhondta/rpl-attacks/blob/master/core/commands.py#L706) is then executed.
Apparently, you modified the repository of Contiki OS and it complains about these changes. I will check this soon for a solution.

yakoubi5103, 2020-08-12:
thank you for this

closed 2020-09-09 by dhondta