dhondta/rpl-attacks #56: Is it possible to edit the malicious and root code?

deGhost, 2020-09-08:
Hello! Thanks for allowing the use of your framework.
Is is possible to modify the code of the z1 motes? If yes, how?

Thank you!

dhondta, 2020-09-09:
Hi @deGhost 

Of course, it is possible by tuning your campaign JSON file, like explained [here](https://rpl-attacks.readthedocs.io/en/latest/campaigns/).

In essence, when you define your JSON dictionary, you can define a key for your experiment that holds a subdictionary whose keys are "`simulation`" and "`malicious`". You can then link a C code for your mote using the following subkeys :

- `simulation` > `root`
- `simulation` > `sensor`
- `malicious` > `type`

The value of these subkeys will resolve to a file path like so : `[framework_folder]/templates/experiment/motes/[type]-[value].c`

So, for instance : 

```js
{
    "my-experiment": {
        "simulation": {
            "root": "test1",
            "sensor": "test2"
        },
        "malicious": {
            "type": "test3"
        }
    }
}
```

Will resolve to the following sources : 

- `[framework_folder]/templates/experiment/motes/root-test1.c`
- `[framework_folder]/templates/experiment/motes/sensor-test2.c`
- `[framework_folder]/templates/experiment/motes/malicious-test3.c`

So beware that you must name your files accordingly.

deGhost, 2020-09-09:
Thanks for taking the time to reply, I checked the c code for each mote. Do you think the z1 mote in simulation can store a machine learning model size= 151KB?

dhondta, 2020-09-17:
Hi @deGhost 
I think it may work but I'm not sure at all. I remember that I struggled with the size limit of 96KB of the MSP430 architecture when it came to testing the build to a real mote. However, as a simulation with Cooja, it could eventually work.
You should definitely Google something like "`contiki z1 mote increase size limit`" and read some forum discussions about this to get a better answer. Note that, if you find something more precise, I would be delighted to get the solution here on this issue.

closed 2020-09-26 by dhondta