dhondta/rpl-attacks #62: Implementing attacks in Cooja

dhondta, 2021-04-09:
respected sir,
I'm from Gujarat technological university in Ahmedabad Gujarat.
I studied for a master's in cybersecurity last semester.
I want to implement DDOS Attack in the cooja simulator in Contiki OS.
I have no idea about how to write code to implement an attack on a
cooja simulator
and how to find the packet delivery ratio and the threshold for a particular
network. how to find the threshold for a particular network.
If possible then please let me know for further discussion.

_Originally posted by @truptipithva in https://github.com/dhondta/rpl-attacks/issues/35#issuecomment-816603104_

dhondta, 2021-04-09:
Hi @truptipithva
Simulating attacks with [Contiki-OS](https://github.com/contiki-os/contiki) and [Cooja](https://anrg.usc.edu/contiki/index.php/Cooja_Simulator) is templated in the [RPL Attacks Framework](https://github.com/dhondta/rpl-attacks), meaning that you [generate (campaigns of) simulations](https://github.com/dhondta/rpl-attacks/blob/master/doc/campaigns.md) from a [predefined template](https://github.com/dhondta/rpl-attacks/tree/master/templates/experiment) to compare a situation of a WSN with and without your attack, relying on [building blocks](https://github.com/dhondta/rpl-attacks/blob/master/templates/building-blocks.json) you can define for tweaking [ContikiRPL](https://github.com/contiki-os/contiki/tree/master/core/net/rpl).
So, for implementing a DDoS attack, you need first to know how you need to tweak RPL so that it will have the desired effect. Then you can add a building block and include it in your campaign file to adapt the simulation with your attack. This is explained [here](https://github.com/dhondta/rpl-attacks/blob/master/doc/README.md).

closed 2021-05-06 by dhondta

azeemjahangir, 2021-07-06:
I need black hole and version number of attacks c file.

sousouk, 2023-03-11:
hello can you help me I have contiki3.0 cooja simulator  and I want to create flooding attack based on DIS message and DIO message please help me