dhondta/rpl-attacks #63: Sinkhole attacks

sanjeetk26, 2021-04-09:
I'm relatively new to simulating WSN attacks. This project seems to be a lifeline. I'm trying to simulate Sinkhole attacks and I wrote the json file according to the first example given in the [tutorial](https://rpl-attacks.readthedocs.io/en/latest/campaigns/) of creating campaigns. However, I am getting this warning:
WARNING Parameter [malicious -> building-blocks] 'sinkhole' does not exist (removed)

On seeing the file building-blocks.json I saw that "sinkhole" wasn't there. But my simulation still ran. The malicious node also did not behave like a sinkhole attack malicious node in the gif file that was created. Did it behave like an ordinary node?

I'm guessing that I would have to add a building-block of "sinkhole" to the  building-block.json file. Being relatively new and not knowing which functions to call etc.. could you please provide me with the code for the building block of the malicious node in sinkhole attacks.

I really need help at this point. Have been trying for quite sometime.

dhondta, 2021-04-10:
Hi @sanjeetk26 

First, thank you for trying this project. As you already noticed, it provides you mechanics to implement and test RPL attacks by comparing situations with and without a malicious mote.

Your reasoning is completely correct ; up to now, you generated a [campaign JSON file](https://github.com/dhondta/rpl-attacks/blob/master/templates/experiments.json) for generating [simulations](https://github.com/dhondta/rpl-attacks/tree/master/templates/experiment) that relies on a [building block](https://github.com/dhondta/rpl-attacks/blob/master/templates/building-blocks.json) that does not exist yet.

This is where the research steps in ; I hoped contributors would write new [building blocks](https://github.com/dhondta/rpl-attacks/blob/master/templates/building-blocks.json) by studying how to implement attacks in [ContikiRPL](https://github.com/contiki-os/contiki/tree/master/core/net/rpl), just like my teammate @bahmadh and me did in our very first research (whose [report](https://github.com/dhondta/rpl-attacks/blob/master/doc/report.pdf) is provided in [supporting documents](https://github.com/dhondta/rpl-attacks/blob/master/doc)). Feel free to add your new attack to the framework too !

**Conclusion** : There is currently no building block for the *Sinkhole Attack*, it still needs to be created.

closed 2021-05-08 by dhondta

far-voyager21, 2025-02-18:
Can you provide the c code which you used for the sinkhole attacker node.