dhondta/rpl-attacks #64: Running Simulations

sanjeetk26, 2021-05-02:
Hi. Great work done here. However, I need some help.

I'm running into 2 issues:

1. If I set the duration of the simulation to about 3000 seconds, it works fine. However, if I set it higher, say 7200 seconds, the simulation runs forever and does not stop. I have tried this with multiple simulations and various times (above 5000 seconds), and the same issue is occurring. I have kept this simulation running for about 5 hours but it still does not end. Can't seem to figure out the problem.

2. On running the simulations for decreased rank attack, I am getting the following histogram for power tracking:
[image]

I can't seem to find much difference between the graphs with and without the malicious node. According to some research papers that I have read, the power consumption graphs look way different from this. This is giving me a hint that maybe the I am implementing the simulation wrongly?

Also, just a small doubt, what is the difference between malicious sensor node and a malicious root node in terms of decreased 
rank or sinkhole attacks?

Any help would be appreciated.

dhondta, 2021-05-04:
Hi @sanjeetk26 
Thank you for posting this.
For your first point, I guess it relates to the Cooja simulator, as the framework only templates simulations to be run on it.
For your second point, I will try to inspect it as soon as possible. Could you post the simulation campaign JSON file you used to include your building blocks ?

For your extra question, the classical [`sensor` template](https://github.com/dhondta/rpl-attacks/blob/master/templates/experiment/motes/malicious-sensor.c) has only its port UDP/3000 open and the [`root` one](https://github.com/dhondta/rpl-attacks/blob/master/templates/experiment/motes/malicious-root.c) can trigger creating the DAG.

sanjeetk26, 2021-05-05:
Hi. Thanks for the reply.

Here's the code that I used:

```JSON
{
  /*  
   All simulations will have the same topology with 10 motes of platform Z1, spread in an square of 200.0m side, with
    a communication range of 50.0m and will last 120 seconds.  
   */
  "BASE": {
    "simulation": {
      "number-motes": 10,
      "target": "z1",
      "duration": 5000,
      "minimum-distance-from-root": 20,
      "wsn-generation-algorithm": "quadrants"
    }
  },
  "dc-att8": {
    "simulation": {
      "title": "Test simulation",
      "goal": "Create a new simulation",
      "root": "echo"
    },
    "malicious": {
      "type": "sensor",
      "building-blocks": [
        "decreased-rank"
      ]
    }
  }
}
```

dhondta, 2021-05-06:
Please check the [academic report](https://github.com/dhondta/rpl-attacks/blob/master/doc/report.pdf) ;
- *Decreased Rank* is part of the *Traffic*-impacting attacks.
- As you can see in the example in page 12/14, the malicious mote advertises a better rank to get included in the WSN just like for a Man-In-The-Middle attack.
- Therefore, it is logical that you see no difference in the power consumption as this attack is not aimed to exhaust power.

closed 2021-05-12 by dhondta