dhondta/rpl-attacks #65: Black hole and decrease rank attacks

azeemjahangir, 2021-07-06:
I need Black hole and version number of attacks code in c file

dhondta, 2021-07-06:
Hi @azeemjahangir 

> **TLDR** ; this source code is made at runtime while compiling a simulation. It can either be retrieved during this operation or regenerated by reproducing the operations from the [building-blocks](https://github.com/dhondta/rpl-attacks/blob/master/templates/building-blocks.json) manually.

While making a [campaign of simulations](https://rpl-attacks.readthedocs.io/en/latest/campaigns/#creation-process) (e.g. [the demo campaign](https://github.com/dhondta/rpl-attacks#demonstration)) or a [single simulation](https://rpl-attacks.readthedocs.io/en/latest/commands/#simulation), the ContikiRPL library is modified based on the selected [building-blocks](https://github.com/dhondta/rpl-attacks/blob/master/templates/building-blocks.json) (using the [available operations](https://rpl-attacks.readthedocs.io/en/latest/building-blocks/), e.g. pattern replacement). The malicious mote is then compiled from its C file, therefore using the modified ContikiRPL library and afterwards immediately removed.
So, if you want to get the modified code, you can simply manually reproduce the modifying operations from the [building-blocks](https://github.com/dhondta/rpl-attacks/blob/master/templates/building-blocks.json) or interrupt the simulation while it is being made, before the process removes the temporary sources for the modified ContikiRPL library.

azeemjahangir, 2021-07-23:
Can any have decrease rank and blackhole in file so i can have an idea for implementation

azeemjahangir, 2021-07-23:
Like with energy consumption of every nodes included

dhondta, 2021-07-23:
Please read the [documentation about building blocks](https://rpl-attacks.readthedocs.io/en/latest/building-blocks/). This will help you understand what you are searching for.

## Decreased Rank Building Block

Here are the transformations (from the [building blocks](https://github.com/dhondta/rpl-attacks/blob/master/templates/building-blocks.json)) for the *decreased-rank* building block :

```json
{
    "RPL_CONF_MIN_HOPRANKINC": 0,
    "rpl-private.h": [
      ["#define RPL_MAX_RANKINC             (7 * RPL_MIN_HOPRANKINC)", "#define RPL_MAX_RANKINC 0"],
      ["#define INFINITE_RANK                   0xffff", "#define INFINITE_RANK 256"]
    ],
    "rpl-timers.c": ["rpl_recalculate_ranks();", null]
}
```

This means : 

- constant `RPL_CONF_MIN_HOPRANKINC` set (line created) in [`contiki/core/net/rpl/rpl-conf.h`](https://github.com/contiki-os/contiki/blob/master/core/net/rpl/rpl-conf.h)
- lines [176](https://github.com/contiki-os/contiki/blob/master/core/net/rpl/rpl-private.h#L176) and [190](https://github.com/contiki-os/contiki/blob/master/core/net/rpl/rpl-private.h#L190) replaced in [`contiki/core/net/rpl/rpl-private.h`](https://github.com/contiki-os/contiki/blob/master/core/net/rpl/rpl-private.h)
- line [95](https://github.com/contiki-os/contiki/blob/master/core/net/rpl/rpl-timers.c#L95) removed in [`contiki/core/net/rpl/rpl-timers.c`](https://github.com/contiki-os/contiki/blob/master/core/net/rpl/rpl-timers.c)

## Blackhole Attack

As it can be seen in the [demo campaign](https://github.com/dhondta/rpl-attacks/blob/master/examples/rpl-attacks.json#L58), the blackhole attack is the combination of the *decreased-rank* building block (see above) and a *drop-messages* one that does not exist yet (you can propose a PR, if relevant :) ).

Johnbucky, 2021-07-23:
This codes might be helpful for you.
[Blackhole-and D-Attacks.zip](https://github.com/dhondta/rpl-attacks/files/6871102/Blackhole-and.D-Attacks.zip)

dhondta, 2021-12-03:
Hi there,

I realize I forgot to respond. My apologies...
Thank you very much ! Indeed, it could be of help, but I don't spend time
anymore to this project, at least for the time being as I'm too busy with
other matters.
Did you mind integrating it into the framework ?

Best regards,

Alex

UUMLM, 2022-11-28:
> This codes might be helpful for you. [Blackhole-and D-Attacks.zip](https://github.com/dhondta/rpl-attacks/files/6871102/Blackhole-and.D-Attacks.zip)

Any read me file or instructions for this code?

researchrinki25, 2024-03-01:
sir can you help me to tell about how to implement attacker code in cooja 3.0

Laileet, 2024-03-09:
Just follow the instructions and you will have the attack files included

Layla Mohammad