dhondta/rpl-attacks #70: I need a help : how to detect rank attack ?

Zouheyr32, 2022-05-31:
Hi every body I need a help ,I use contiki 3.0 os cooja simulator but I'have difficult to detect rank attack , knowing  I use standard deviation  method .and the code  attack is running .

l'm waiting anouther solution and positive answre.

dhondta, 2022-06-20:
Hi @Zouheyr32 !
Can you provide a bit more information ?
What do you mean with the standard deviation method ?

sousouk, 2023-03-10:
hi  can you help me I work in contiki 3.0 cooja simulator and I want to create a flooding attack based on : DIS message and DIO message please help me and thank you

dhondta, 2023-03-11:
@sousouk If you want to discuss about the flood attack, you can open a new issue. This one relates to another one.
You can find more information in my [academical report](https://raw.githubusercontent.com/dhondta/rpl-attacks/master/doc/report.pdf) ; the hello flood attack is already implemented in RPL Attacks Framework and can be tested by using the [demonstration](https://github.com/dhondta/rpl-attacks#sunglasses-demonstration).

sousouk, 2023-03-13:
Ok but what about DIS flooding attacks how create??