dhondta/rpl-attacks #72: vagrant up command

UUMLM, 2022-12-04:
following the listed commands i have the below error appears after vagrant up command:
The box 'dhondta/rpl-attacks' could not be found or
could not be accessed in the remote catalog. If this is a private
box on HashiCorp's Vagrant Cloud, please verify you're logged in via
`vagrant login`. Also, please double-check the name. The expanded
URL and error message are shown below:

URL: ["https://vagrantcloud.com/dhondta/rpl-attacks"]
Error: Could not resolve host: vagrantcloud.com

UUMLM, 2022-12-04:
I followed the instructions Murat Ugur KIRAZ introduced in his blog https://en.mukiraz.com/2022/06/dhondts-rpl-framework/
it solved the issue

closed 2022-12-04 by UUMLM

dhondta, 2022-12-04:
Hi @UUMLM !
Thank you for reporting this.
I see you got "`Error: Could not resolve host: vagrantcloud.com`". This is likely to be the cause of `vagrant up` failing. The box is indeed available at this URL.
Glad you found an article helping.