dhondta/rpl-attacks #74: DIS flooding

sousouk, 2023-03-14:
hello , I want to create DIS flooding attack but I want to change in file rpl-icmp6.c  to do this attack can you give me solution

sousouk, 2023-03-14:
can I create DIS flooding attack in file rpl-icmp6.c ? please help I need this

dhondta, 2023-03-14:
Hi @sousouk 
A priori, I guess it could be the right way. This is the hard part, you need to test ; this involves making experiments with a custom external RPL library the running the simulation to verify if it gives what you expect.
The work method is as follows : 
1. Copy the `core/net/rpl` library of Contiki to a new folder where you will apply your modifications
2. In the framework, while `make`-ing your simulation, enter `ext_lib=path/to/your/core/net/rpl` (see the documentation [here](https://rpl-attacks.readthedocs.io/en/latest/commands))at the end to set this particular parameter
3. Run your simulation and see if the results are what you expect
4. If not, go back to 2.

sousouk, 2023-03-31:
Hello . I want to do DIO attack I want to change the interval to send
message dio but in rpl-icmp6.c  can you help me and thank you

sousouk, 2023-04-01:
hello can you help me , how I use the perl script in contiki cooja simulator  please help me I need this