dhondta/rpl-attacks #76: Question about power tracking

Phantom-bot7, 2023-08-13:
Hi, I am confused about the power tracking graph, could you please explain the what each item means in power tracking graph?
[image]

TX: ?
RX: ?
ON: ?
INT: ?
And also, how to generate this chart via original powertracker data? 

Thanks!

dhondta, 2023-09-18:
Hi @Phantom-bot7 !
Sorry for the late response.
As of the [code of Cooja](https://github.com/contiki-os/contiki/blob/master/tools/cooja/apps/powertracker/java/PowerTracker.java#L268C22-L268C22), the following nomenclature is used for power consumption : 
- *ON* : the *mote* is on, in idle state
- *TX* : the *mote* was *transmitting*
- *RX* : the *mote* was *receiving*
- *INT* : the *mote* was in *interfered* state
As you can see on your plot, mote 3 has about 7% of its power consumed while idle and about 6% while receiving. Same applies to mote 7 and 10. I guess that your malicious mote (11) was performing a hello flood and that motes 3, 7 and 10 were in its vicinity.

closed 2023-10-07 by dhondta

Phantom-bot7, 2023-11-14:
@dhondta  sorry for late reply, thanks for your answer, it helps me a lot.